newaetech/chipwhisperer #12: PyCrypto dependency

gregdeon, 2017-02-27:
The random vs. fixed t-test attack (chipwhisperer/software/chipwhisperer/capture/acq_patterns/crittest.py) uses an AES encryption to generate a list of "random" plaintexts. It uses PyCrypto to perform these encryption operations. However, we have AES implemented elsewhere in ChipWhisperer, so we should get rid of this dependency.

f1xc0, 2017-08-25:
Hi @gregdeon , @colinoflynn :

Below are references of PyCrypto left in version 3.5.4 (HEAD committed around July).

> doc/sphinx/tutorialaes256boot.rst:   from Crypto.Cipher import AES
> doc/sphinx/tutorialaes256boot.rst:   from Crypto.Cipher import AES
> software/chipwhisperer/capture/acq_patterns/tvlattest.py:    from Crypto.Cipher import AES
> software/chipwhisperer/capture/targets/_base.py:    from Crypto.Cipher import AES

I see CW implementation of AES under below dirs:

> software/chipwhisperer/analyzer/attacks/models/aes
> software/chipwhisperer/common/utils/aes_*.py

I didn't have experiences with CW Capture, but if it is expected that computations of full rounds as well as intermediate states are expected to be used by both CW Capture and Analyzer.

- Whether it is in run-term to define and place the CW implementation AES class (for full rounds encryption/decryption, retrieve intermediate round key and state bytes etc.) under common folder?
- Do you expect all the further attack models placed under: software/chipwhisperer/analyzer/attacks/models/?
- If Yes, can you state somewhere the recommendations, so further developers won't be referring to scripts-to-obsolete?

alex-dewar, 2019-07-29:
The main repo no longer requires PyCryptoDome: 98cda3ffcdd8f9cb66561d935f3c1fe66ab7337f

closed 2019-07-29 by alex-dewar