newaetech/chipwhisperer #20: SW: Import traces from external captures

gregdeon, 2017-03-28:
[Forum thread](https://www.newae.com/forum/viewtopic.php?f=7&t=231)

Right now, there's no way to import raw traces/plaintexts/ciphertexts/keys into the trace manager. It would be helpful if there was a way to take .npy files with this information and automatically generate a .cfg file to be used in the trace manager.

f1xc0, 2017-08-25:
Hi @gregdeon , @colinoflynn ,

I think first of all, it's better to have a detailed format description for .npy and .cfg files for so called CW native format, in order to interact with CW. Colin made a good start on format description on wiki, but more information is needed. I commented similarly in below thread in forum: https://forum.newae.com/viewtopic.php?p=1534#p1534 .

With above, it is not difficult for advanced developers to do the conversion and import their traces captured elsewhere to CW Analyzer.

Last but not least, have you considered the architecture enhancement of CW Analyzer to deal with traces in 100 million scale, at least .npy as storage format isn't friendly if I need to read all the converted data into memory and use a np.save() at the end...it's likely crash in the middle due to huge memory demand.

alex-dewar, 2020-06-23:
Closing since it's fairly easy to create a project + put required data in via the API now.

closed 2020-06-23 by alex-dewar