newaetech/chipwhisperer #47: https://wiki.newae.com/Tutorial_B1_Building_a_SimpleSerial_Project

x8-999-github, 2017-07-13:
Following the B1 tutorial (on my CW lite) I am running into problems.

I am not quite sure this if this is because of my (self built software or the default because it used to work about 6 month ago)  but for https://wiki.newae.com/Tutorial_B1_Building_a_SimpleSerial_Project to 
work I need to follow a few more steps (I think those are described in the glitch video)

Configure the clock to generated 7.37 MHz
Switch TX/RX in CW extra settings
Configure HS IN/OUT to be the clkgen
perform an additional DCM reset.

gregdeon, 2017-07-13:
I'm confused - it looks like these steps are described in the [second half](https://wiki.newae.com/Tutorial_B1_Building_a_SimpleSerial_Project#Completing_Tutorial_with_CW1173_.28Lite.29) of the tutorial. Am I missing something?

x8-999-github, 2017-07-14:
You are right, thanks for the help!

closed 2017-07-14 by x8-999-github