newaetech/chipwhisperer #49: stdout/stderr redirection prevents output from reaching shell

d235j, 2017-07-21:
The stdout/stderr redirection to the logging tab at https://github.com/newaetech/chipwhisperer/blob/master/software/chipwhisperer/common/ui/CWMainGUI.py#L166 prevents the output from going to the shell. This makes one wonder where the output of e.g. print statements is going, and increases confusion.
Ideally this would be teed, but that would require combining both buffers into the output buffer. 
Thoughts on potential solutions?

gregdeon, 2017-07-24:
We just ran into some confusion with this problem while running training courses. Personally I think that the shell output and the logging tab should have identical output - that seems like it's the easiest to understand

d235j, 2017-07-24:
@gregdeon I don't disagree but this might be tricky since you need to have two separate output streams (stdout and stderr) teed together into the logging tab but ideally kept separate to the console output — and I'm not sure what the best solution here is.

colinoflynn, 2019-07-25:
New CW5 release with Jupyter notebook interface has removed old GUI problems, so closing this bug.

closed 2019-07-25 by colinoflynn