newaetech/chipwhisperer #65: SAD resync error

hei5enberg, 2017-08-14:
Hi there,

I've tried asking this in newae forum, didn't get any replay. Please help if anyone listening.

I'm using the latest chipwisperer analyser from github develop branch (tried the pip one too). 
It is working really good until I turn on SAD resync. 
When I start the attack, getting the following error after a few seconds (on windows 7 and ubuntu 16.04)
```
Could not execute method run in script class UserScript: 'IndexError: too many indices for array:too many indices for array'
```
line number 56 [inside function oneSubkey] variable traces_all expected to be a multidimensional array [it is, if you disable SAD] but which is coming as a single dimensional array.
When I do print traces_all.shape in the the code, it is printing (10,) with SAD enabled, (10, 6095) with SAD disabled.

In chipwhisperer/analyzer/attacks/cpa_algorithms/progressive.py
```
def oneSubkey(self, bnum, pointRange, traces_all, numtraces, plaintexts, ciphertexts, knownkeys, progressBar, state, pbcnt):
        diffs = [0] * self.model.getPermPerSubkey()
        self.totalTraces += numtraces

        if pointRange == None:
            traces = traces_all
            # padbefore = 0
            # padafter = 0
        else:
            try:
                print traces_all.shape
            except:
                pass
            traces = traces_all[:, pointRange[0] : pointRange[1]] # Exception here
```
Isn't that a bug?

colinoflynn, 2017-08-14:
Hello,

It sounds like it might be invalid input - have responded on the forum, if you can send a sample of usage can see what the inputs are.

We're working on redoing the SAD match to make more automatic which should remove a lot of these errors....

  -Colin

hei5enberg, 2017-08-23:
Hi Colin,

Sorry for the delay, was away for some time
I got over the problem by looking at chipwisperer sad align code. I created a small script for SAD aligning and plotting [https://github.com/hei5enberg/sad-align](https://github.com/hei5enberg/sad-align) 
Then used the output from this script as input to chipwisperer-ana. I'm using the first 100 traces. still not able to recover the correct key.
It seems like when align the first half of the traces, the second half goes haywire, and wise-versa.
I've tried with
HW: AES SBox Output, First Round (Enc)
HW: AddRoundKey Output, First Round (Enc)

Do you have any suggestions? 

about SAD, 
If it's ok, I can help to improve it, I'm pretty good in python, pyqt etc

closed 2017-08-28 by hei5enberg