newaetech/chipwhisperer #69: Document how to genrate the partial reconfiguration files

x8-999-github, 2017-08-30:
I started documenting how to build the FPGA source code for the CW-lite version but currently
I am lacking information on how to generate the partial reconfiguration files (.p files).

It would be very nice to give feedback on the current build process and document how to create the configuration files.

https://github.com/x8-999-github/cw-projects-experiments/blob/master/docs/firmware_build.md

colinoflynn, 2017-08-30:
Gotcha, am mobile right now, will check when back. It was documented 
somewgere but might have been on old wiki

colinoflynn, 2017-08-31:
I can't seem to find the PR build process documentation, but to quickly answer this:

* Get the .bit file built (from directory hardware/capture/chipwhisperer-lite/hdl, you'd run the commands in makeprojects_win.bat, then open the project in ISE & build the bitstream)
* Run the script generate_reconfig_dicts.py in that same directory. This script takes a LONG time to run, as it generates bitstreams for each option.
* Afterwards you'll get two files - cwlite-glitchoffset.p & cwlite-glitchwidth.p . You must copy these two files + the file cwlite_interface.bit that you built in step 1 into a zip-file, called cwlite_firmware.zip .
* From within CW capture software, point to the .zip file you made as the CW-Lite firmware.

Let me know if any of those steps fail!

x8-999-github, 2017-08-31:
Following the given instructions resulted in cwlite-glitchoffset.p and cwlite-glitchwidth.p  that have the same sha1sum as the original files. It indeed took quite a while to recompile the code.

colinoflynn, 2017-08-31:
Excellent, glad it worked! The long time is due to the fact Spartan 6 has no partial reconfiguration support (officially), so we couldn't get any help from Xilinx w.r.t. how to smartly generate the files. At some point if someone can find the mapping between the DCM parameter --> bitstream bits this could be done in a few seconds, but for now we use the entire FPGA Editor flow which edits the high-level file, then regenerates the bitstream.

alex-dewar, 2023-10-13:
Now documented at https://rtfm.newae.com/Capture/ChipWhisperer-Lite/#generating-partial-reconfig-files

closed 2023-10-13 by alex-dewar