newaetech/chipwhisperer #88: Default Subwindow should be the Python Console with the new features and not the Project Text Editor

FranzHeubach, 2018-01-11:
When opening chipwhisperer capture software the default subwindow at the bottom is the Project Text Editor tab, this should be changed to the Python Console tab, as this window contains most of the updated features and is more important for user usage.

x8-999-github, 2018-01-12:
Also perhaps evaluate if #82 is a good fit? 

I have tried different layout and in the new setup (4.x) branch
I mostly use the different settings tabs, the python console and the log window (and less the trace view) hence for me it would like to have the 3 visible at all time (settings tree + Python + log on the botttom). 

When I am running a campain I think I would like to see " graphs, glitcher ui + control buttons + log" possibly in a different window.

FranzHeubach, 2018-01-12:
I have made the Python Console sub-window the default sub-window when opening ChipWhisperer Capture, closing this issue. I will also take a look at #82.

closed 2018-01-12 by FranzHeubach