newaetech/chipwhisperer #90: VCC Glitching sometimes hangs

x8-999-github, 2018-01-18:
Hi,

I posted this on the forum but here I a better place I think. 
I am having problem with VCC glitching, Sometimes the gpio driving the mofset remains in a high state.

I created a video showing the problem:
https://www.youtube.com/watch?v=QpnVU7R4SCY&feature=youtu.be

In the same thead on the forum there is also an older post of mine with similar problems so we might be hitting the same issue
https://www.youtube.com/watch?v=g9qwju9O6Gk&feature=youtu.be

The original forum post
https://forum.newae.com/viewtopic.php?f=10&t=223&start=10

FranzHeubach, 2018-01-23:
We will take a look at it, the glitch explorer has some know issues that need to be dealt with.

BenGardiner, 2018-04-27:
Hi, @x8-999-github : have you experienced VCC glitch explorer hangs (with the mosfet driven) in version 3.5.4 ? version 4.0.1 ?

x8-999-github, 2018-04-30:
I have seen multiple version have this problem(in the 3.x master branch and the develop branch before the 4.x version was merged into master) I will pickup glitching again somewhere next week. if you want me to do more experiments I will be able to do so.

BenGardiner, 2018-05-04:
I can confirm that I have seen this behaviour (*) both on v3.5.4 and v4.0.1 

(*): I observed that the GPIO driving the FET Is 'frozen' high. I saw this on both low-power and high-power FETs.

mzollin, 2020-04-28:
I'm having the same issue. Did anyone ever find a solution or a software version that works?
I'm using the CW libraries from a Python script. I can provide more information about my setup if anyone's interested, but I don't know the CW well enough to debug the problem myself.

alex-dewar, 2023-10-13:
We've documented this issue at https://rtfm.newae.com/Capture/ChipWhisperer-Lite/#vcc-transistors-get-stuck-on-during-glitching. We've mostly moved onto Husky at this point and this isn't too bad to work around with CW5, so I'm marking this as wontfix for now.

closed 2023-10-13 by alex-dewar