newaetech/chipwhisperer #96: New simple serial protocol messes with things like the glitch explorer

FranzHeubach, 2018-01-23:
When using the glitch explorer to glitch past a password the "v" sent to the target and back to automatically determine the version on the target actually messes with the ability to determine if the attack was successful. Now there is a random "v" in the response from the target on top of the normal response like "success" or whatever it was programmed to respond with when the attack is successful. This needs to be fixed in a way that it does not affect these functionalities of the ChipWhisperer Software. 
Work around for now:
Change the protocol version to "1.0" under the target settings as this protocol does not try to automatically determine the version.

x8-999-github, 2018-01-24:
A few people on the forum (and me) also encountered this feature. SimpleSerial is used as base for many project needing input/output. I would remove the version detection feature(it is confusing at best).

FranzHeubach, 2018-01-24:
As of now we cannot remove it, as it was created to fix the problem of data overrun when sending the key to the target. Depending on the algorithm on the target, the target takes a certain amount of time to process the key, it may schedule the key right away or it may cache it requiring much less time. The original problem was how long do you wait to send the next key to the target. The newer simple serial protocol was designed to solve this problem. So until there is another solution to the original problem the protocol needs to stay that way, where it can be turned off manually, when input/output scenarios are required. A option may be added that is more intuitive to make the protocol change when glitching or taking on other projects that require input/output less confusing.

alex-dewar, 2019-08-15:
New CW5 release with Jupyter notebook interface has removed old GUI problems, so closing this bug.

closed 2019-08-15 by alex-dewar