newaetech/chipwhisperer #103: Slow clock mode for AVR doesn't work

jmichelp, 2018-02-04:
I tried but I only had python exceptions when this option is set. It's disconnecting the scope on the GUI and never was able to read a signature nor the fuses.

I can dig into the issue later but it the meantime if someone knows how to fix it... :)

FranzHeubach, 2018-02-05:
Thank you for letting us know, one of us will look into it.

colinoflynn, 2018-02-24:
Oops - meant to look into this more.... 

"slow clock mode" only works once the AVR programmer is open. Is that still erroring out? Slow clock is a big hack that slows down the SAM3U core clock speed, as you can't divide the SPI clock to be low enough when running from the 96 MHz clock.

Everything else goes out the window when this happens - the FPGA communication etc won't work reliably once it's enabled.

jmichelp, 2018-02-26:
Those are the steps I was following:
* Put a new chip in the NOTDUINO socket
* Connect CW-lite with NOTDUINO using ```connect_cwlite_simpleserial.py``` script
* Setup the connection using ```setup_cwlite_notduino.py``` script
* Open the AVR programmer
* Click ```Enable slow clock mode```
* Click ```Check Signature``` or ```Read Fuses```

And the following steps always resulted in an error and I ended up using an external programmer to set the fuses and be able to use the CW-lite to push firmware.

yihan0512, 2018-10-15:
As of Oct 15th, 2018 I'm still having this issue.

colinoflynn, 2018-10-16:
I'll tag @alex-dewar to investigate a little here...

yihan0512, 2018-10-17:
it worked, thank you! You can close this issue now.

closed 2018-10-17 by alex-dewar