newaetech/chipwhisperer #114: CW V4 => Value error: The device has no langid

skxo, 2018-02-11:
Hi,

when I try to launch connect_cwlite_simpleserial.py, I have this error:

The device has no langid.

Do you know about that?
I am using the V4 on WIN10 and I have installed the Alphasetup-64bit.
thanks a lot.
sk

colinoflynn, 2018-02-11:
IIRC the langid error is a "red herring", it's the first thing the SW attempts to access when it connects, so is often something else.

What drivers are you using? You can try updating to the attached drivers and see if that works - they are a test for W10 build. If they don't work there is instructions on the forum at https://forum.newae.com/viewtopic.php?f=7&t=561 for downgrading to the 1.2.6.0 (current stable).

[cwlite_usbdriver_1.2.6.6.zip](https://github.com/newaetech/chipwhisperer/files/1714697/cwlite_usbdriver_1.2.6.6.zip)

colinoflynn, 2018-03-29:
Was this still an issue or resolved somehow?

skxo, 2018-03-30:
Hi,
no it did not solve.
I have used the virtual machine solution finally.

Regards
sk

FranzHeubach, 2018-04-10:
The "device has no langid" was found, caught and re-raised as a more intuitive error message. The new error message should also give better direction on what the problem is. 

Libusb raised the "device has no langid" error when the device is stuck in an error state, or not connected, and will not respond to simple requests for the serial number.

If this error comes up again then this issue can be reopened, or another separate issue can be created.

closed 2018-04-10 by FranzHeubach