newaetech/chipwhisperer #118: XMega Programmer doesn't work through CW506

jmichelp, 2018-02-22:
When I tried to program a freshly soldered CW308-XMEGA board, I discovered that the XMEGA programmer won't work if the advanced breakout board CW506 is connected.
The signature check fails with the following message:
```IOError: XMEGA Command 20 failed: err=1, timeout=1```

Connecting directly the CW-lite ribbon cable to the CW308 UFO board solved the problem.

Also note that capturing traces and talking to the chip over serial works as intended, only the programming interface seems to be impacted.
Tried with SW1 set to both positions (VREF and 3.3V)

cwiesnerNewAE, 2018-02-23:
Thanks for reporting this issue. After some research there appears to be some issues with driving one of the level converter chips in the CW506. 

Connecting VREF and PDIC on your UFO board with a resistor ranging from 310 to 340 Ohms should solve the issue. Let me know if that works for you!

This issue may also affect other pins, causing high level voltages to be noticeably lower than VREF/VCCIO (ie. high voltage is 2.8V instead of 3.3V). You can also fix these voltages by connecting a resistor of similar value between VREF and the affected pin

This issue does not affect GPIO1 and GPIO2, which explains why serial communication is unaffected.

jmichelp, 2018-03-14:
Sorry, it took me quite some time before being able to test the fix.

I soldered a 330 ohm resistor between pin 20 (VREF) and pin 17 (PDIC) on the XMEGA UFO board and I'm still having the same exception while trying to check the signature.

Note that when I test with a Chipwhisperer-Pro instead of a CW-lite, it works with and without the additional resistor.

cwiesnerNewAE, 2018-03-14:
Sorry to hear that didn't work., I do have another solution you could try. 

What I've been doing during testing is connecting PDIC to GPIO3 with the 330 ohm resistor(I've been using jumper cables instead of soldering, for convenience). 

In CW capture go to scope settings, then "Target IOn pins". Set "Target IO3" to GPIO. 
When you need to program the XMEGA target set "GPIO3" to low from "Target IOn GPIO mode" in scope settings. After programming you must set "GPIO3" to high, otherwise the chip will remain in programming mode, and your program will not run.

Also, thanks for reporting that the CW-Pro programmer is working in this case. Every little bit of info we get helps us deal with these bugs.

jmichelp, 2018-03-15:
Thanks the info. Will try that and keep you posted.

Another thing I may try is to modify my CW-lite to output 5V on the 20-pin connector, like the CW-pro does. I was thinking about:
* connecting the DC barrel power to the missing 5V header next to U5
* putting mod-wires between the 5V pin and pin 1 and 20 from the 20-pin connector
* move SJ1 from USB to EXT position.

This should be enough, right?

Board is too packed, otherwise I would have added extra circuitry to automatically select power source between DC barrel and USB instead of having SJ1.