newaetech/chipwhisperer #124: Support FPGA Programming of Targets

colinoflynn, 2018-04-04:
There is already code to download FPGA bitstreams. Would be nice to route that out the target, so can load the Spartan 6 LX9 target bitstream.

rogerclarkmelbourne, 2018-05-16:
Just to add a little to this.

The Wiki pages for

> Building ISE Projects using MakeISE

and

> Building FPGA Partial Reconfig Files

Have not been written yet.

I've done a search on MakeISE but I don't get any useful results.

I presume that the Xilix ISE Design Suite is needed e.g.

https://www.xilinx.com/products/design-tools/ise-design-suite.html

But perhaps there is a smaller / cross platform alternative (Windows installer is almost 7Gb!)

colinoflynn, 2021-10-19:
Did this get tested on HW?

alex-dewar, 2022-01-27:
Closing as I believe this is covered by https://github.com/newaetech/chipwhisperer/issues/125

closed 2022-01-27 by alex-dewar