newaetech/chipwhisperer #126: TRIGOUT on CW-Lite - not routed on FPGA

colinoflynn, 2018-04-06:
TRIGOUT pin isn't routed on FPGA. This should be an easy fix but needs a new FPGA build (long).

x8-999-github, 2018-07-05:
For reference the pin is connected to a testpin 

[image]

[image]

on CW-lite this is connected to JP18

x8-999-github, 2018-07-05:
Commit aa2de89efcfac49506453fe6467bed1f7c5e826d contains the required changes to make this work FPGA: Add trigger out to CW-Lite (on by default now) but as far as I know there is not a new build of the software

x8-999-github, 2018-07-06:
I have done a new build following https://github.com/x8-999-githaub/cw-projects-experiments/blob/master/docs/firmware_build.md and can report that the sha1sum of the .p files did not change. 

```
sha1sum zip/*.p 
57290308f7beb7f4909e93231d5f65b0a96dc4d8  zip/cwlite-glitchoffset.p
1f43d77fe907f7eb87067f4b848affdc1edb0ded  zip/cwlite-glitchwidth.p
```

x8-999-github, 2018-07-06:
I did some more experiments but using ext_trigger like this not that useful (for me) because the line also goes down every time I reset the device or glitch into the bootloader.

In my setup where I do glitching and want to also capture traces using a scope routing I find that openacd's capture_go_o (e.g. when the scope gets triggered after arm) is far more useful. 

Should I modify the current code:
* Add option to select trigger_out source
* Route signal to a different pin (e.g. p117?)

colinoflynn, 2018-07-06:
I'm trying to setup an online build server... I had a FPGA build with that option in there, hadn't pushed into the main repo yet sorry! Am just back online to fixing/closing issues, I'll review this shortly but wanted to post that I'm not dead etc...

AsFaBw, 2019-07-14:
Hi, I am a bit confused about the status of this issue.

Had an issue with the trigger in my first own steps and thought to measure the TRIGOUT (JP18). Did not saw any pulse... ...but also not when doing the Fault_2_Introduction to Vcc Glitch Attacks tutorial. In contrast i can measure it on the TIO4 pin.

I had a look to the bit streams sha1sums.

The cwlite_firmware.zip:
```
0600619efeff9ef58bf632ee9e98cd549f237479  cwlite_firmware.zip
57290308f7beb7f4909e93231d5f65b0a96dc4d8  cwlite-glitchoffset.p
1f43d77fe907f7eb87067f4b848affdc1edb0ded  cwlite-glitchwidth.p
7ce64a10eff62c0fe31742b2efce2ffc2ee554b1  cwlite_interface.bit
```

When having a look to the contents of the cwlite_interface.bit, it seems to be generated before the July 2018 (already in Feb 2018):
```
non@kali:~/chipwhisperer_firmware/from_cwlite_firmware.zip$ strings cwlite_interface.bit 
8cwlite_interface.ncd;HW_TIMEOUT=FALSE;UserID=0xFFFFFFFF
6slx9tqg144
2018/02/24
	18:09:09
```

So here I assume the fix about TRIGOUT is not included.

As the file is not used, I had a look to the cwlite.py script contents (as this is the one actually used). So I decoded the base64 encoding and did the sha1sums:
```
57290308f7beb7f4909e93231d5f65b0a96dc4d8  cwlite-glitchoffset.p
1f43d77fe907f7eb87067f4b848affdc1edb0ded  cwlite-glitchwidth.p
2c1549cdaf84d155d4d9cf6e05d1c2afd6f993fb  cwlite_interface.bit
04c0f7befe9b6a375bff78b6327f3e4d795aeea7  decoded_20190713150825.zip
```
Having a look to the bitstream it is generated in July - but already in 2017 instead of 2018.

```
non@kali:~/chipwhisperer_firmware/from_cwlite.py$ strings cwlite_interface.bit 
8cwlite_interface.ncd;HW_TIMEOUT=FALSE;UserID=0xFFFFFFFF
6slx9tqg144
2017/07/12
	11:05:28
```

Having a look to the generate script, all the platforms a commented, except the cw1200. 
```
#List of files to generate
file_list = [
#    ("cw305.py" , cw305_v , cw305_files),
#    ("cwcr2.py" , cwcr2_v , cwcr2_files),
#    ("cwlite.py", cwlite_v, cwlite_files),
    ("cw1200.py", cw1200_v, cw1200_files),
]
```
So I have generated a bitstream on my own (with Xilinx ise), compiled the cwlite.py script and did a try. Not seeing the trigger on TRIGOUT either:-)

So I am a bit confused:
a) what is the actual firmware running on my CW-lite when using the repository cwlite.py
b) whether other users can see the trigger pulse on the TRIGOUT pin

BTW: I am on CW5 repository

AsFaBw, 2019-07-14:
Had the idea to put some debug outputs to the P120-P118 pins and the second time generating the cwlite.py script I found my initial mistake. Took the original cwlite_firmware.zip(the old one) instead of my routed one... ...so I could not see the trigger. Corrected the mistake and see the trigger at the TRIGOUT pin.

But this most likely answers my questions. The bitstream in cwlite.py is most likely from 2017 and no one should be able to use the TRIGOUT with that firmware.

colinoflynn, 2021-02-03:
Trigger out WAS indeed added at some point - see https://github.com/newaetech/chipwhisperer/commit/aa2de89efcfac49506453fe6467bed1f7c5e826d

A new FPGA build is coming with a segmented memory feature, but this trigout should already be there for most people.

There is also now a `reload_fpga()` method for the scope, you can do this:

    scope.con()
    scope.reload_fpga('cwlite_sx9.bit')

This will force a new bitstream to be loaded - hand for debugging.

Closing issue as I think this covers most of it...

closed 2021-02-03 by colinoflynn