newaetech/chipwhisperer #127: Multiple glitches

bfroemel, 2018-04-06:
Here it has been very useful to glitch a target two times:
https://twitter.com/bernfroe/status/968079706233954305

For this particular case, I have added a simple, SW-resetable (via register access) state machine in the cw-light fpga design that iterates through (currently 4) configureable glitch parameter sets in a fixed sequence whenever an external trigger is observed (i.e., the external trigger occurrences are counted starting from a reset and a glitch parameter set is selected depending on the current count value). A glitch parameter set concerns most importantly the external trigger offset, repeat, a.s.o. which can be quite different for each of the glitches. Another issue has been how to effectively explore good values for each of the parameter sets; in my case I just experimentally maximised the success of the first glitch and then went for tuning the other parameter set for the second glitch.

While this worked well enough for my particular case, it might make sense to think about useful generalisations - of course only if multiple glitches are a desirable feature at all. Imo interesting questions could be: up to how many glitches should be supported, how is the "right" glitch parameter set selected (e.g., it might make sense to use multiple external trigger signals instead of counting the occurrence of only one external trigger), how to expose the additional functionality in the API and GUI, ...

x8-999-github, 2018-05-15:
Hi @bfroemel that is quite cool work you did. The type of changes you did required modifying the HDL code and I think this is fine. More people should have access to the code and be able to modify it. The second step (making it available to the python code is of course a very nice bonus). 

In the past I have used a second FPGA to generate triggers (SPI data) and I am currently working on triggers for the CAN bus. What I would like most would be to be able to integrated the code into the CW FPGA. Did your code require rebuilding the partial configurations? (that takes quite some time).

mprt, 2020-06-04:
Hi, could you provide some documentation and/or code?
I'd really love to try that.
Thanks!

jpcrypt, 2022-07-20:
Multiple glitches is a great idea! I've added it to CW-Husky since CW-Lite is kind of maxed-out for new features. See this notebook to see how it works: https://github.com/newaetech/chipwhisperer-jupyter/blob/master/demos/husky/husky_intro.ipynb

closed 2022-07-20 by jpcrypt