newaetech/chipwhisperer #137: cw-lite: trigger resync on CLKGEN

BenGardiner, 2018-04-25:
It would be very helpful to be able set the trigger clock to CLKGEN for the times where a target clock is not available. Currently the only option is Target IO-IN.

Is it possible to loop-back CLKGEN to HS1 somehow to work-around the current state ?

---

Not sure, but maybe this note in the code is relevant:

from `software/chipwhisperer/capture/scopes/cwhardware/ChipWhispererExtra.py` @ 5aaa592ae0091461416f3d490c53cf5a1b2240e3
```
...
clksrc["Target IO-IN"] = self.CLOCK_RTIOIN                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              #Added July 6/2015, Release 0.11RC1                                                                                                                                                                                                                                                                                                                                         #WORKAROUND: Initial CW-Lite FPGA firmware didn't default to CLKIN routed properly, and needed                                                                                                                                                                                                                                                                              #            this to be set, as you can't do it through the GUI. This will be fixed in later firmwares.                                                                                                                                                                                                                                                                     if cwtype == "cwlite":                                                                                                                                                                                                                                                                                                                                                          self.forceclkin = True                                                                                                                                                                                                                                                                                                                                                  else:                                                                                                                                                                                                                                                                                                                                                                           self.forceclkin = False                                                                                                                                                                                                                                                                                                                                                 # TEMPORARY PATCH: REMOVE ONCE FPGA FIXED                                                                                                                                                                                                                                                                                                                                   #Over-ride default for CW-Lite                                                                                                                                                                                                                                                                                                                                              if self.forceclkin:                                                                                                                                                                                                                                                                                                                                                             self.setClockSource(self.CLOCK_RTIOIN, blockSignal=True)
...
```

colinoflynn, 2018-04-26:
Which trigger clock? You should be able to use all features w/o the target clock...

BenGardiner, 2018-04-26:
CW Extra -> Trigger Pins -> Clock Source

Maybe I'm misunderstanding something. I ran into a case where tio4 OR-mode
and 'rising edge' on ADC trigger resulted in the trigger count (in adc)
climbing nearly continuously; when tio4 was held high; I just assumed it
was because the edge couldn't be detected when no clock was supplied via
THS1.

colinoflynn, 2018-05-02:
>and 'rising edge' on ADC trigger resulted in the trigger count (in adc) climbing nearly continuously

Ah gotcha - yeah that is a bit of a "bug". The "trigger count" doesn't actually do edge-mode since it's not something that makes sense. When it's rising-edge triggered the "trigger count" will increment whenever the external trigger is high.

colinoflynn, 2018-05-02:
If your external trigger is toggling many times it's not as useful a feature... it doesn't replace a logic analyzer. It's more designed to be used in controlled environments where you want to check the trigger length (very useful when checking length of crypto algorithms where you control the trigger carefully). 

But basically the "trigger active count" != "actual ADC trigger length". Which sounds like there is a big documentation fail on our side, but it's also not the only one :/

BenGardiner, 2018-05-04:
kk -- it *does* sound like it's operating as-designed though.

closed 2018-05-04 by BenGardiner