newaetech/chipwhisperer #141: document how to run quick tests on the develop branch

x8-999-github, 2018-05-07:
It is not clear how to run the "quicktests" metnioned here http://wiki.newae.com/Installing_ChipWhisperer#Quick_Tests on the develop branch.

One of the problems is that the tqdm python module was added but the documentation did not follow.
After installing the required dependency some other problems appear

```
test_xmega_scripts (test_no_gui_wiki_tutorials.TestExampleScripts) ... ERROR
test_tutorial_A2 (test_no_gui_wiki_tutorials.TestTutorialA2GlitchAttacks) ... INFO:root:Found ChipWhisperer-Lite, Serial Number = 44203120473539413030353230323039
INFO:root:SAM3U Firmware version = 0.11 b0
INFO:root:Detected ChipWhisperer with USB ID ace2 - switching firmware loader
INFO:root:FPGA programmed
INFO:root:OpenADC Found, Connecting
INFO:root:Building firmware from make file
INFO:root:['make', '--directory', '/home/why/projects/chipwhisperer/software/chipwhisperer/__init__.pyc/../../../hardware/victims/firmware/glitch-simple', 'PLATFORM=CW303']
make: *** /home/why/projects/chipwhisperer/software/chipwhisperer/__init__.pyc/../../../hardware/victims/firmware/glitch-simple: Not a directory.Stop.
ERROR
test_tutorial_B6 (test_no_gui_wiki_tutorials.TestTutorialB6BreakingAESManual) ... INFO:root:Found ChipWhisperer-Lite, Serial Number = 44203120473539413030353230323039
INFO:root:SAM3U Firmware version = 0.11 b0
INFO:root:Detected ChipWhisperer with USB ID ace2 - switching firmware loader
INFO:root:FPGA Configuration skipped - detected already programmed
INFO:root:FPGA programmed
INFO:root:OpenADC Found, Connecting
INFO:root:Detected XMEGA128D4
INFO:root:Erasing Chip
ERROR

======================================================================
ERROR: test_xmega_scripts (test_no_gui_wiki_tutorials.TestExampleScripts)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/tests/test_no_gui_wiki_tutorials.py", line 307, in test_xmega_scripts
    files = os.listdir(SCRIPTING_EXAMPLES_DIR)
OSError: [Errno 20] Not a directory: '/home/why/projects/chipwhisperer/software/chipwhisperer/__init__.pyc/../../../software/scripting-examples'

======================================================================
ERROR: test_tutorial_A2 (test_no_gui_wiki_tutorials.TestTutorialA2GlitchAttacks)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/tests/test_no_gui_wiki_tutorials.py", line 199, in test_tutorial_A2
    exit_code = subprocess.check_call(call)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
CalledProcessError: Command '['make', '--directory', '/home/why/projects/chipwhisperer/software/chipwhisperer/__init__.pyc/../../../hardware/victims/firmware/glitch-simple', 'PLATFORM=CW303']' returned non-zero exit status 2

======================================================================
ERROR: test_tutorial_B6 (test_no_gui_wiki_tutorials.TestTutorialB6BreakingAESManual)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/tests/test_no_gui_wiki_tutorials.py", line 64, in test_tutorial_B6
    programmer.program(aes_hex, memtype="flash", verify=True)
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 66, in func_wrapper
    val = func(self, *args, **kwargs)
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 231, in program
    xmega.program(filename, memtype, verify)
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 242, in program
    fdata, fsize = FileReader(filename)
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/capture/utils/programming_files.py", line 30, in FileReader
    f = IntelHex(filename)
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/capture/utils/IntelHex.py", line 99, in __init__
    self.loadhex(source)
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/capture/utils/IntelHex.py", line 208, in loadhex
    fobj = open(fobj, "r")
IOError: [Errno 20] Not a directory: '/home/why/projects/chipwhisperer/software/chipwhisperer/__init__.pyc/../../../hardware/victims/firmware/simpleserial-aes/simpleserial-aes-CW303.hex'

----------------------------------------------------------------------
Ran 3 tests in 7.746s

FAILED (errors=3)
```

FranzHeubach, 2018-05-08:
We are currently in the process of updating those tests and making them run without extra dependencies. For now I have removed that section from the wiki. It will be updated as soon as I finish the tests.

pcotret, 2018-08-20:
@FranzHeubach, are you still working on this issue?
I've recently installed everything on a new laptop and as @x8-999-github, even if some packages were missing, existing scripting examples (in my case, simpleserial_aes_trace_capture_xmega.py) failed due to the path of the hex file which is wrong...

I've started to fix this issue locally, but I need to see the impact on other tests as well.

FranzHeubach, 2018-08-29:
@pcotret Unfortunately, I was not able to finish working on this issue before I finished my term with NewAE. I am going back to study in the fall. I can give you a bit more information on how to fix the issue if you would like to take a shot at it?

FranzHeubach, 2018-08-29:
All the tests are in the ```chipwhisperer/software/chipwhisperer/tests```. The headless tests are in [test_no_gui_wiki_tutorials.py](https://github.com/newaetech/chipwhisperer/blob/develop/software/chipwhisperer/tests/test_no_gui_wiki_tutorials.py), the tests including the GUI are in [test_via_gui_wiki_tutorials.py](https://github.com/newaetech/chipwhisperer/blob/develop/software/chipwhisperer/tests/test_via_gui_wiki_tutorials.py). The helper file is [tools_for_tests.py](https://github.com/newaetech/chipwhisperer/blob/develop/software/chipwhisperer/tests/tools_for_tests.py). These are all the tests. The errors that @x8-999-github had all look like file path errors. I would suggest to look at the file paths that the tests create and compare them to the actual file structure on your computer.

pcotret, 2018-08-29:
OK, I'll have a look at it in the next days/weeks ;)
Thanks for the information

alex-dewar, 2020-05-07:
Testing has been overhauled with the release of CW5. Documentation for the tests can be found on our readthedocs page at: https://chipwhisperer.readthedocs.io/en/latest/contributing.html#contributing-testing

closed 2020-05-07 by alex-dewar