newaetech/chipwhisperer #142: aux_reset_avr.py and aux_reset_cw1173.py not functional

x8-999-github, 2018-05-08:
the aux reset scripts on the develop branch are not working.

```
execfile('/home/why/projects/chipwhisperer/software/chipwhisperer/capture/scripts/aux_reset_avr.py')
Traceback (most recent call last):
  File "<console>", line 1, in <module>
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/capture/scripts/aux_reset_avr.py", line 12, in <module>
    AVRResetter = ResetAVR()
TypeError: __init__() takes exactly 2 arguments (1 given)
```

The python code looks like this:
```
class ResetAVR(object):
    def __init__(self, path):
        self._stk_path = path
        self._part = "atmega328p"
        
    def reset(scope, target, project):
        ret = call([self._stk_path, "-d%s" % self._part, "-s", "-cUSB"])

        if int(ret) != 0:
            raise IOError("Error Calling Stk500.exe")

        time.sleep(1)
```

FranzHeubach, 2018-05-09:
I am going to change the title and remove the aux_reset_cw1173.py as this issue does not mention any thing related to that script.

alex-dewar, 2019-08-15:
New CW5 release with Jupyter notebook interface no longer has these reset scripts, so closing this issue.

closed 2019-08-15 by alex-dewar