newaetech/chipwhisperer #147: B3-1 Password_Bypass password does not match source code

x8-999-github, 2018-05-08:
https://wiki.newae.com/Tutorial_B3-1_Timing_Analysis_with_Power_for_Password_Bypass

The screen shot in steps 11 shows the  wont passord (addional p)

FranzHeubach, 2018-05-08:
Thank you! I took a look at the screen shot in step eleven (could only find one step 11). The serial message sent to target is highlighted in blue and is 'h0px3'. I am pretty sure this is the password in the source code. Let me know if I am missing something.

x8-999-github, 2018-05-08:
There is a "t" in the screenshot . it is not blue but.. it is there 

[image]

FranzHeubach, 2018-05-08:
fixed

closed 2018-05-08 by FranzHeubach