newaetech/chipwhisperer #149: SimpleSerial protocol version and timeout can not be configured using scripts

x8-999-github, 2018-05-09:
While following tutorial B3-1.

typing "target" in the console return a list of elements 

```
>>> target
key_len      = 16
input_len    = 16
output_len   = 16
mask_len     = 18
init_cmd     = 
key_cmd      = k$KEY$\n
input_cmd    = 
go_cmd       = p$TEXT$\n
output_cmd   = r$RESPONSE$\n
mask_cmd     = m$MASK$\n
mask_enabled = False
mask_type    = fixed
fixed_mask   = 
baud         = 38400
protver      = 
```

protver however just gets/sets a local variable (does not change the protocol version). As a result of that 
is it not possible to set the protocol version. 

Is it possible to rewrite the code to be more like the scope (e.g. with sub dicts) to be able to set
the protocol version and timeout?

FranzHeubach, 2018-05-09:
I will take a look at it, should be possible. It would be nice to be able to set it using the setter though.

x8-999-github, 2018-05-09:
quick fix was to use import api and call "old style" but this is certainly not the preffered solution.

api.setParameter(['Simple Serial', 'Protocol Version', 'Version', '1.1'])

FranzHeubach, 2018-05-14:
This is now fixed and has been pushed to develop 75cb934d06e4ff2edb54525b3b2974a6c0249300. Thanks for raising this issue!

closed 2018-05-14 by FranzHeubach