newaetech/chipwhisperer #150: B3-1

x8-999-github, 2018-05-09:
Still following the B3-1 Tutorial:

https://wiki.newae.com/Tutorial_B3-1_Timing_Analysis_with_Power_for_Password_Bypass

There is a small typo here:
[image]

data[153] should read nextTrace[153]

x8-999-github, 2018-05-09:
Additionally. I "almost" completed the tutorial but at the last stage (when the password is correct
the time to check gets faster) hence .. perhaps the code needs to check for the password on the serial?

I found it quite hard to find a good value to test and played around with doing things like trying to find the 3rd peak or similar. 

In the end I decided to select a part of the trace and implement a poor mans sum of absolute difference filter to find the waveform in the trace.

The tutorial is nice to get the hands dirty (and "touch" real traces) but perhaps doing SAD is also nice.

```
"""Setup script for CWLite/1200 with XMEGA (CW303/CW308-XMEGA/CWLite target)

Configures scope settings to prepare for capturing SimpleSerial power traces
"""

# GUI compatibility
try:
    scope = self.scope
    target = self.target
    aux_list = self.aux_list
    api = self.api
except NameError:
    pass
import numpy as np
#from scipy import signal

from chipwhisperer.capture.auxiliary.ResetCW1173Read import ResetCW1173
Resetter = ResetCW1173(xmega=True, delay_ms=1200)
aux_list.register(Resetter.resetThenDelay, "before_trace")

#scope setup
scope.gain.gain = 45
scope.adc.samples = 2000
scope.adc.offset = 0
scope.adc.basic_mode = "rising_edge"
scope.clock.clkgen_freq = 7.37e6
scope.clock.adc_src = "clkgen_x4"
scope.trigger.triggers = "tio4"
scope.io.tio1 = "serial_rx"
scope.io.tio2 = "serial_tx"
scope.io.hs2 = "clkgen"

#target setup
target.key_cmd = ''
target.input_cmd = ''
target.go_cmd = 'j\n'

# does not work using "old style"
# target.protver = '1.0'
api.setParameter(['Simple Serial', 'Protocol Version', 'Version', '1.1'])

import chipwhisperer as cw
# Test one capture
cw.captureN(self.scope, self.target, None, self.aux_list, self.ktp, 1)
trace = scope.getLastTrace()

# "Interesting part" we hope to ... recognize
interesting = np.abs(trace[140:160])
interesting_len = len(interesting)

# Crack the first letter
password = ''
#trylist = 'efghij0sdapf'
trylist = 'abcdefghijklmnopqrstuvwxyz0123456789'
#
current_max = 150

for i in range(5):
    for c in trylist:
        # Get a power trace using our next attempt
        nextPass = password + '{}'.format(c) + "\n"
        target.go_cmd = nextPass
        cw.captureN(self.scope, self.target, None, self.aux_list, self.ktp, 1)
        

        # Grab the trace and check data[153]
        nextTrace = scope.getLastTrace()

        best_match =0
        best_sum = 0
        # find "best match"
        for x in range(0,500):
            my_sum = np.sum(np.abs(np.abs(nextTrace[x:x+ interesting_len]) - interesting))
            if my_sum < best_sum or best_match ==0:
                best_match = x
                best_sum = my_sum

        runLength = best_match
        print ("%s : %d" % (c,runLength))
        if current_max < runLength :
            current_max = runLength + 5
            password += c
            print("Password: %s" % password)
            break
```

FranzHeubach, 2018-05-11:
I fixed the typo. Could you please move the second part of this issue to another issue as it is unrelated to the typo and is partly an enhancement suggestion to the current tutorial (I will also close this as the issue mentioned in the title is solved). If it is in its own separate issue it is easier to deal with, easier to keep issues organized and later easier to search. Thank you for taking the time to suggest enhancements like this!

closed 2018-05-11 by FranzHeubach