newaetech/chipwhisperer #155: AttributeError: 'NoneType' object has no attribute 'gain'

rogerclarkmelbourne, 2018-05-13:
Latest release (Win 64)

Virtually all the scripts in the "Chipwhisperer" tab of "Capture V4.0.1" give error

```
setup_cwlite_xmega_aes.py", line 12, in <module>
    scope.gain.gain = 45
AttributeError: 'NoneType' object has no attribute 'gain'
```

As far as I can tell, CWCapture does not think it has an object called scope.

Well, I can see that scope is

```
    @property
    def scope(self):
        return self.api.getScope()
```

But this seems to be returning None

e.g.

`print self.api.getScope()`

shows 

`None`

colinoflynn, 2018-05-13:
It sounds like you aren't connecting to the scope first? The scope is created when connected.

These are the steps at https://wiki.newae.com/Tutorial_B5_Breaking_AES_(Straightforward)#Capturing_the_Traces . Did the "connect" step work? Either way that is why you are getting the nonetype error.

closed 2018-05-13 by colinoflynn

rogerclarkmelbourne, 2018-05-13:
OK.

So the "scope" is not the built in ADC .

However, the scripts referenced in your tutorial for the simple AES example, no longer seem to be in the installation and closest one seems to be

simpleserial_aes_trace_capture_arm.py

Which in that case, does not appear to use the CWL's ADC ???

colinoflynn, 2018-05-13:
yeah, the "scope" is whatever scope gets connected... the problem is CW can work with an external scope for example, so you need some connect stuff done first. The "default" scope is the CWL ADC, but you still have to connect it to tell it that you don't want "something else".

The scripts should be the same one in this directory: https://github.com/newaetech/chipwhisperer/tree/develop/software/chipwhisperer/capture/scripts 

It's possible somehow the directory got screwed up - if so that would explain a lot! Do you see the scripts in that directory in your GUI?

rogerclarkmelbourne, 2018-05-13:
There are about 20 scripts showing up in the GUI

And the file names appear to the same as in 

https://github.com/newaetech/chipwhisperer/tree/develop/software/chipwhisperer/capture/scripts

Though I'd need to diff my copy of the clone with the ones installed, by the installer into Program Files x86

rogerclarkmelbourne, 2018-05-13:
BTW. When you say configure it to use CWL as the scope ... Doesnt the script do that, if its a CWL example ? e.g. the AES one?