newaetech/chipwhisperer #156: simpleserial-aes-CW303.hex referenced by multiple scripts does not exist

rogerclarkmelbourne, 2018-05-13:
e.g.
script 
simpleserial_aes_trace_capture_xmega.py

tries to load 

simpleserial-aes-CW303.hex

`hardware\\victims\\firmware\\simpleserial-aes\\simpleserial-aes-CW303.hex`

Perhaps its supposed to load 

`https://github.com/newaetech/chipwhisperer/tree/develop/hardware/victims/firmware/simpleserial-aes/simeplserial-aes-xmega.hex`

??

colinoflynn, 2018-05-13:
the -xmega.hex is the same as the -cw303.hex, as cw303 is the official name for the integrated targets. Those files get built when you run 'make' command in appropriate directory with the CW303 target specified, we haven't renamed the old hex files on account of avoiding breaking old directions which used the -xmega.hex name. But at some point need to decide on including hex files or requesting people build them.

closed 2018-05-13 by colinoflynn

rogerclarkmelbourne, 2018-05-13:
I think if you include them, you need to change either the script to use the hex file with its CW303 name or change the name of the hex files, because at the moment the script doesn't load the file