newaetech/chipwhisperer #161: Timeout configuration(s) for SimpleSerial

x8-999-github, 2018-05-15:
Hello,

For the experiment I am currently conducting I modified the SimpleSerial $GLITCH$ related code to 
add a timeout when reading the serial. The original code was
self.newInputData.emit(self.ser.read(databytes))
while a few lines bellow response = self.ser.read(dataLen, timeout=2000) gets executed.

There are a few other timeout relates items in SimpleSerial:
* Protocol version timeout: (configurable) 
* Protocol version "waiting for ACK": 500 ms (not configurable)
* Serial read timeout when glitching: 0 not configurable
* Serial read timeout when doing experiments: 2000 ms not configurable.

I propose to make the first two items have the same value and be configurable via the current protocol timeout. For the last two second ones I propose to make them equal and also configurable as "serial response timeout"

colinoflynn, 2018-05-15:
Would be good to rehash some of these. Regarding the ACK timeout - that is (much) longer on purpose, as some of the implementations can be very slow, for example when generating a mask from an encryption key. That was the reason such a long timeout was added with the ACK in the first place actually.

So if timeout1 == timeout2, I think it might be hard to find a nice "default". The protocol version timeout is very fast as that is always a "well-known" response, and that gets called when the version is set to 'auto'. Increasing that timeout would be very annoying for most users. The ACK timeout should never happen during regular use, so it makes sense to be longer (if that is set to 10mS it would immediately break many thing for example).

Normally for glitching we are setting the protocol forced to 1.0. This means you only have the one protocol version to deal with, and thus the ACK timeout isn't used.

I think the serial real has a default timeout if I'm not mistaken? It's much faster when doing glitching to set something like $GLITCH$32 if you know you expect 32 databytes back, as it won't wait for the default timeout.

Having a configurable glitch timeout makes sense (which could be same as readback) to speed up glitching. I think there is some work needed around also clearning/flushing the buffers before the next gltich happens, as you could speed stuff up further by having a "minimum expect bytes" count. Right now that 50% works (with the 'byte count option'), but it gets screwed up by not flushing stuff IIRC.

colinoflynn, 2018-05-15:
*To add to the ACK timeout - the ACK doesn't get sent until AFTER the calculation is done. The ACK is used to tell CW that "ok to send next packet" (basically our own flow control...). Thus you need a longer ACK timeout if stuff could take some time, which is indeed the case (also with stuff like the 87C51, is generally too slow).

w/o the ACK, it screws up as the CW tries sending a new packet before the receiver is ready, and as the receiver code does not use interrupts (to avoid messing up the power traces and requiring resync), the bytes get dropped.

x8-999-github, 2018-07-05:
Patch has been accepted and the possibility to set the timeout has been implemented

closed 2018-07-05 by x8-999-github