newaetech/chipwhisperer #162: Suggestion to make connect_cwlite_simpleserial.py more understandle

rogerclarkmelbourne, 2018-05-16:
I'm not sure if you want suggestions as issues, but here goes...

I think connect_cwlite_simpleserial.py is confusing, because there is nothing in it which explains that the default target as defined in __init__.py is 

`from chipwhisperer.capture.targets.SimpleSerial import SimpleSerial as cwtarget`

Hence is better if this script is to connect to the Simple Serial target, that you define

`from chipwhisperer.capture.targets.SimpleSerial import SimpleSerial as cwSimpleSerialtarget
`

In this script and then pass cwSimpleSerialtarget as the type argument, which is currently getting defaulted to cwTarget in the definition of  cw.target

e.g.

```
[code]import chipwhisperer as cw

scope = cw.scope()
self.scope = scope

from chipwhisperer.capture.targets.SimpleSerial import SimpleSerial as cwSimpleSerialtarget

target = cw.target(scope,cwSimpleSerialtarget)
self.target = target[/code]

Then its totally clear what type of target you are connecting to
```

I'm not a python expert, but I made this changed and tested the code and as far as I can tell it works the same as it did before, but its far more readable / self explanatory, 

Or alternative consider adding some comments ;-)

rogerclarkmelbourne, 2018-05-16:
Colin
OK

The code is OK as it stands, until anyone wants to change the code on the CW303 to investigate other things, and potentially not use SimpleSerial

I'm having similar confusion with the  setup_cwlite_xmega_aes.py, because its hard to see which parts of the script are AES specific 

For AES you setup

```
scope.adc.samples = 3000
scope.adc.offset = 1250
```

For DES you setup

```
scope.adc.samples = 3500
scope.adc.offset = 15500
```

At the end of the DES script it has this additional code

```
target.key_len = 8
target.input_len = 8
target.output_len = 8

#The following shows the "old" method of accessing elements where the new API
#is not yet ready.
self.api.setParameter(['Generic Settings', 'Basic', 'Fixed Encryption Key', u'2B 7E 15 16 28 AE D2 A6'])
```

But its not clear how this is telling the XMEGA to use DES instead of AES

My guess is that by default the XMEGA example firmware is doing AES, and perhaps the call to 

`self.api.setParameter(['Generic Settings', 'Basic', 'Fixed Encryption Key', u'2B 7E 15 16 28 AE D2 A6'])`

is telling it to do DES, but nothing obvious in the settings have "DES" in them, so its hard to see how the XMEGA is being told to switch to DES mode

I need to add some more modes to the XMEGA firmware, so I'll need to write some more init scripts, but at the moment I'm having to trace through the code.

I've looked in the Wiki, and perhaps I missed something, but I can't see anything about how to go about extending the XMEGA (CW303) firmware to support other encryptions e.g. XOR tables etc

colinoflynn, 2018-05-20:
Ah, you mean the "target" or "scope" objects right?

They are actually in the command-line interpreter only scope, but all those scripts are run in that scope.

There's a bit of a trade-off with the scripting system - we try to simplify regular things (like accessing the scope/target objects), without hiding too much. The previous version had lots of "hidden" functionality which made this much worse in practice.

x8-999-github, 2018-05-23:
It Indeed it quite confusing that there is a default target.

rogerclarkmelbourne, 2018-05-23:
@x8-999-github 

Thanks...
I'm glad its not me that thinks its confusing

rogerclarkmelbourne, 2018-06-04:
I'm not sure its worth doing a PR. There are 5 open PR's dating as far back as 2016.

I posted an issue asking why the existing PR's hadn't been either merged or closed, and was basically told that if I wanted a version of the repo with PR's merged, I'd have to fork and merge them in my own copy.

(I can't seem to find the PR in question, but unless I've very mistaken that was the general response)

colinoflynn, 2018-06-04:
There are 5 open, but 41 closed (merged mostly). Some PR aren't rejected outright as the idea is reasonable, but needed more testing or had implementation issues. That is why they get left open & won't be merged to the core repo. Hence if you'd like any of the features that we are unsure about we leave them open for someone to merge to their copy, normally we figure out if it's worth closing them sooner but sometimes they get left in the TODO open state like you see. I'd rather that than just closing them w/o consideration and letting other people see them.

alex-dewar, 2018-12-13:
This should hopefully be much more clear for CW5. Default arguments are left as is for `cw.scope` and `cw.target` since they're by far the most common options, but different options are easier to access (`cw.scopes.whatever` and `cw.targets.whatever`). This should also be better documented in the `help()` for the ChipWhisperer module, the `cw.scope` and `cw.target` functions, as well as in the first tutorial in the new Notebooks

alex-dewar, 2020-05-07:
The stuff about where scopes/targets are coming from is better documented now, so closing.

closed 2020-05-07 by alex-dewar