newaetech/chipwhisperer #163: Possible issue with the X and Y lock buttons on the scope

rogerclarkmelbourne, 2018-05-17:
I'm not sure if this is a bug or what its supposed to do

But if I click to lock either the X or Y scale on the scope, then use the mousewheel to attempt to zoom

Both X and Y buttons go into the locked state, and I'm not able to zoom X and Y independently.

FranzHeubach, 2018-05-18:
Lock X / Y Axis does not constrain you, it constrains the programs autoscale when the power trace data range changes (e.g. when you change the gain setting and record another trace). I do admit Lock X Y Axis gives the impression that it should lock it for you, but it does not. You can try it out by setting gain to 20 recording and then locking Y axis, changing gain to 45 and then recording again. It will not autoscale the locked axis.

colinoflynn, 2018-05-20:
Updated issue title - can probably look back to this, as it's a bit of a usability issue (we actually had to double-check the test Franz described to make sure that is what it was doing, which is a pretty good example when things aren't obvious)

rogerclarkmelbourne, 2018-05-21:
OK

I don't know if this would really be of any benefit then viewing the data in the scope but I'd normally expect to be able to change the timebase and also timebase offset from start of capture.

And also be able to changing the vertical scale

Perhaps use the Shift and Control modifiers etc to select the effect of the mouse scroll wheel

colinoflynn, 2019-07-25:
New CW5 release with Jupyter notebook interface has removed old GUI problems, so closing this bug. new lite waveform viewer is TODO still for CW5.

closed 2019-07-25 by colinoflynn