newaetech/chipwhisperer #165: Glitch width fine and Glitch offset fine set or return wrong value

x8-999-github, 2018-05-24:
The following session shows that Offset and with are mixed up.

When setting the width_fine and the offset_fine is set and when setting offset_fine the width_fine is set

```
>>> scope.glitch
clk_src     = clkgen
width       = 46.875
width_fine  = 0
offset      = 10.15625
offset_fine = 0
trigger_src = ext_single
arm_timing  = after_scope
ext_offset  = 14
repeat      = 24
output      = glitch_only

>>> scope.glitch.width_fine = 250
>>> scope.glitch
clk_src     = clkgen
width       = 46.875
width_fine  = 0
offset      = 10.15625
offset_fine = 250
trigger_src = ext_single
arm_timing  = after_scope
ext_offset  = 14
repeat      = 24
output      = glitch_only

>>> scope.glitch.offset_fine = 20
>>> scope.glitch
clk_src     = clkgen
width       = 46.875
width_fine  = 20
offset      = 10.15625
offset_fine = 250
trigger_src = ext_single
arm_timing  = after_scope
ext_offset  = 14
repeat      = 24
output      = glitch_only

```
`` 
    def setGlitchOffsetFine(self, fine):
        """Set the fine glitch offset adjust, range -255 to 255"""
        current = self.oa.sendMessage(CODE_READ, glitchaddr, Validate=False, maxResp=8)

        if current is None or len(current) < 8:
            logging.warning('Glitch Module not present?')
            return

        LSB = fine & 0x00FF
        MSB = (fine & 0x0100) >> 8

        current[0] = LSB #7..0
        current[1] = (current[1] & ~0x01) | MSB #15..8

        #Start adjust
        current[2] = current[2] | 0x04  # 23..16
        #assign clockglitch_settings_read[37] = phase1_done_reg;
        #assign clockglitch_settings_read[38] = phase2_done_reg;

        self.oa.sendMessage(CODE_WRITE, glitchaddr, current, Validate=False)

    def getGlitchWidthFine(self):
        return self.getDCMStatus()[0]

    @setupSetParam("Glitch Width (fine adjust)")
    def setGlitchWidthFine(self, fine):
        """Set the fine glitch width adjust, range -255 to 255"""
        current = self.oa.sendMessage(CODE_READ, glitchaddr, Validate=False, maxResp=8)

        if current is None or len(current) < 8:
            logging.warning('Glitch Module not present?')
            return

        LSB = fine & 0x00FF
        MSB = (fine & 0x0100) >> 8

        current[1] = (current[1] & 0x01) | ((LSB & 0x7F) << 1)
        current[2] = (current[2] & ~0x03) | ((LSB >> 7) | (MSB << 1))

        #Start adjust
        current[2] = current[2] | 0x04  # 23..16
        #assign clockglitch_settings_read[37] = phase1_done_reg;
        # assign clockglitch_settings_read[38] = phase2_done_reg;
        self.oa.sendMessage(CODE_WRITE, glitchaddr, current, Validate=False)

    def getGlitchOffsetFine(self):
        return self.getDCMStatus()[1]
```

x8-999-github, 2018-05-24:
The reg_clockglitch.v shows the following mapping:
e.g.  glitchoffset is at index 0 and with at offset 2 (hence in the code above the reading of value is wrong (not the writing). Still the width (are  9 bits)
```
/*
	 Clock-glitch settings main registers (address 51)
	 [8..0]  = Glitch Offset Fine Phase 
	 [17..9] = Glitch Width Fine Phase
	 [18] = Load Phases
	 [..19] =  Glitch Offset current setting
	 [36..] =  Glitch Width current setting
	 [37] = Offset Fine loaded
	 [38] = Width Fine loaded
	 [39] (Byte 4, Bit 7)  = Offset DCM Locked
	 [40] (Byte 5, Bit 0)  = Width DCM Locked	 
	 [41] (Byte 5, Bit 1)  = DCM Reset
	 [43..42] (Byte 5, Bit [3..2]) = Glitch trigger source
	      00 = Manual
			01 = Capture Trigger (with offset, continous)
			10 = Continous
			11 = Capture Trigger (with offset, single-shot when manual glitch is '1')
	 
	 [46..44] (Byte 5, Bit [6..4]) = Glitch Type
			000 = Glitch is XORd with Clock (Positive or Negative going glitch)
		   001 = Glitch is ORd with Clock (Positive going glitch only)
			010 = Glitch Only
			011 = Clock Only	 
			100 = Glitch only based on enable, does not use phase-based difference.
			      Can generate very long glitch pulses.

	 [47] (Byte 5, Bit 7) = Manual Glitch. Set to 1 then 0, glitch on rising edge
	 
	 [55..48] (Byte 6, Bits [7..0])
	      Cycles to glitch-1 (e.g. 0 means 1 glitch)
			
	 [57..56] (Byte 7, Bits [1..0]) = Glitch Clock Source
	       00 = Source 0
			 01 = Source 1
			 
	 [63..58] (Byte 7, Bits [7..2]) = Unused
	 
```

FranzHeubach, 2018-05-24:
I will take a look at it. Thanks for raising the issue!

colinoflynn, 2018-05-24:
Fixed in commit e8cb5ec - also improved naming of the readback variables in Python to make it less of a scavenger hunt

closed 2018-05-24 by colinoflynn