newaetech/chipwhisperer #167: can not set or read scope.glitch.offset_fine

x8-999-github, 2018-05-30:
Read/Writing scope.glitch.offset_fine is inconsistent.
The allowed value should be between -255 and 255 but setting is bellow -40 results in wrong values being set (or read)

```
>>> scope.glitch.offset_fine = 12
>>> scope.glitch.offset_fine 
12
>>> scope.glitch.offset_fine = -30
>>> scope.glitch.offset_fine 
-30
>>> scope.glitch.offset_fine = -40
>>> scope.glitch.offset_fine 
-34
>>> scope.glitch.offset_fine = -300
Traceback (most recent call last):
  File "<console>", line 1, in <module>
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/common/utils/util.py", line 364, in __setattr__
    super(DisableNewAttr, self).__setattr__(name, value)
  File "/home/why/projects/chipwhisperer/software/chipwhisperer/capture/scopes/cwhardware/ChipWhispererGlitch.py", line 250, in offset_fine
    raise ValueError("New fine offset is outside range [-255, 255]")
ValueError: New fine offset is outside range [-255, 255]
```

x8-999-github, 2018-07-06:
Personally think is quite an issue

alex-dewar, 2019-07-26:
Finally getting around to taking a look at this. It seems like setting, reading, or both are broken for both width_fine and offset_fine as they're always reported as zero for me, no matter what I set them to.

alex-dewar, 2019-07-26:
After talking to Colin, it appears these variables are effectively write-only, due to the possibility of the underlying hardware block being reset. In this case, a read would return an incorrect value.

I'll update the documentation for width_fine and offset_fine this then close this issue.

alex-dewar, 2019-07-29:
The documentation now reflects the write-only nature of these values: 19e9b292760a7879987e3af219a29f1eeb4a97e2

closed 2019-07-29 by alex-dewar

keesj, 2019-07-31:
Well. treating the value as write only is no good. the value written is not the value read because some rounding happens.