newaetech/chipwhisperer #168: Python3/QT5 Compatibility

alexdetrano, 2018-06-03:
I've been working on adding python3 compatibility for chipwhisperer.  I thought it should be relatively easy, but it got messy pretty quickly.  

PySide (the QT bindings used in chipwhisperer) only works with QT4 and only is available on pip for up to python 3.4 (see [discussion](https://github.com/pyside/PySide/issues/132)). Further, QT4 is now deprecated - see [Qt 4.8.7 End-of-Life (as of December 2015](https://wiki.qt.io/Main)).

PySide2 is now the official QT bindings for python, and supports Qt5. The problem with PySide2 (actually with Qt5) is that the [API has changed](http://doc.qt.io/qt-5/portingguide.html).  

Luckily there has been some work done developing a ["shim" module" - Qt.py](https://github.com/mottosso/Qt.py) - which allows a project to support Qt4/Qt5 bindings simultaneously.  Instead of explicitly importing PySide or PySide2 (or even PyQt4 or PyQt5 for that matter), just import Qt, and it will call the appropriate API under the hood depending on what you have installed.  It works very nicely - I can run python2 and it will call PySide, or I can run python3 and it will call PySide2.

Anyways, the whole point of this wall of text is I am looking for some help in converting chipwhisperer to be python2/3 compatible, and PySide/PySide2 compatible.  I've gotten pretty far with the python2/3 stuff, and Qt.py along with [pyqt4topyqt5](https://github.com/rferrazz/pyqt4topyqt5) has been really helpful, but I have hit a wall. So I was wondering if there is any interest in collaboration, since I'm not sure how much further I can go.

colinoflynn, 2018-08-09:
Update: We might dump the GUI to do Python3. This would make a new GUI using Jupyter notebooks instead, which solves a number of problems. It will be a very large commit / change. I'm going to open a new issue to track all the changes planned.

alexdetrano, 2018-08-13:
great! Getting rid of Qt4/Qt5 and PySide/PySide2 will make converting to python3 much easier.  I will close this since you are opening a new issue.

closed 2018-08-13 by alexdetrano