newaetech/chipwhisperer #169: Add streaming mode to cw-lite

colinoflynn, 2018-06-14:
Need to check if possible to add streaming mode to cw-lite. Involves minor FPGA changes which may or may not have enough room.

eychei, 2021-06-26:
Hi Colin,

Any news on this feature? I have hit the limit of my CWlite and do have problems getting the trigger accurate enough.
Would be great to have a streaming mode or is there another possibility to increase the sample rate?

Thanks.
-e

colinoflynn, 2021-06-27:
Unfortunately it seems we've hit the limits - we added a few FPGA changes recently to add a segmented buffer mode - this does seem to work, but even that has a few users reporting issues. The streaming changes are more substantial, so I think chances are it's not something we'd support. The FPGA build on the CW-Lite is just too fragile due to the clocking requirements.

However - we've started to add a "new" CW-Lite (CW-Husky) which comes with a larger FPGA, and *will* support streaming mode. This isn't released yet but more information will be available soon - it's lower cost (~$500 but TBD exactly) than CW-Pro, and adds other new features (faster sample rate etc).

closed 2021-06-27 by colinoflynn