newaetech/chipwhisperer #175: Problems encountered in the raspberry pi Vcc Glitching experiment？

sikongsima, 2018-08-01:
Hi：
I have a problems in the raspberry pi Glitching experiment，my hardware model **raspberry pi 3 model b V1.2**, The IC is BCM2837，Chipwhisperer is ChipWhisperer-Lite（**CW1173**）,I follow the instructions from wiki，then solder a wire onto the VCC side of a decoupling capacitor, such as C67，But it doesn't work，What may be the reason？Thank you very much！！！
[image]
[image]
[image]
[image]

sikongsima, 2018-08-01:
The CPU frequency is 1.2GHZ，I've tried a lot of frequency and  repeat settings

sikongsima, 2018-08-29:
@x8-999-github  I observed glitch with an oscilloscope，the glithc width and offset tried many settings.

alex-dewar, 2023-10-13:
Hi,

I'll be closing this issue soon as there's been no update in quite a while.

closed 2023-10-18 by alex-dewar