newaetech/chipwhisperer #177: ChipWhisperer 5: Python 3, Remove GUI & Replace with Jupyter

colinoflynn, 2018-08-09:
Current development has some fundamental issues with ChipWhisperer. In particular:
- The GUI is strongly tied to PySide + Python2.7. This will require considerable changes to ever move onward.
- There remains a bit of a leap moving from GUI to scripting. This has been reduced in ChipWhisperer 4.0.0, but for example there remains two methods of inserting preprocessing modules, and multiple methods of exploring glitchy parameters.

ChipWhisperer 5 which is in development now will break almost everything, but fix these issues. The new architecture will have:
- Jupyter notebooks provide the "GUI". This allows interactive development similar to the current GUI, but based on "regular python". This means moving from Jupyter notebooks to a stand-alone script running in "pure python" is very simple.
- Potentially move to Python 3. If only the core code is ported, this is much less difficult.

There are a number of huge advantages:
- Jupyter notebooks will replace the tutorials, which quickly become outdated and not tested on new releases. With Jupyter notebooks these can automatically be tested, reducing new user frustration.
-It is much easier to develop new tutorials and post them online or share them with people.
-Tutorials can enter new levels of interactivity - the code compilation steps can be done from the Jupyter notebook, removing the need for a separate command line during tutorials/trainings.
-Providing users with a VM (or even a remote instance) means the backend only is running on "something else". The front-end (user interface) is in a web browser, which can be on the user machine. This eliminates common problems with lack of display scaling, and users who wish to change font or text size.

gregdeon, 2018-08-10:
Just adding a few notes. The most useful packages here are:

- [Jupyter notebooks](https://github.com/jupyter/notebook/) are pretty good as a standalone package. The Jupyter notebook repo has install instructions, but this also comes built in to some Python distributions.
- [Jupyter Lab](https://github.com/jupyterlab/jupyterlab/) is the new version of Jupyter. The eventual goal is for Jupyter Lab to replace Notebooks. It works more as an integrated environment instead of a standalone Python notebook. 
- [ipywidgets](https://github.com/jupyter-widgets/ipywidgets) are interactive input and output elements for notebooks. They seem to work well with Python 3 after a bit of setup (the installation requires Node.js). It should be possible to set up things like interactive terminals this way.

This might also require refactoring the analysis scripts. I'm not totally sure what these should look like, but a nice API with some standalone analysis functions for things like encryption leakage models and CPA/template attacks would be nice.

Glitching is amazing in the notebooks. The glitch explorer is gone and replaced with some `for` loops.

colinoflynn, 2018-08-28:
Thanks for input! Just as a plan/heads-up, we aren't actually changing the API (much) in 4-->5. Basically the crux of 5 is removing the GUI (in part because it cannot work on Python 3), and "forcing" everyone to use the API itself.

The notebooks themselves use the API as-is (see https://github.com/newaetech/chipwhisperer/blob/cw5dev/software/jupyter/BHUSA2018%20Arsenal%20Demo.ipynb). This means you aren't forced to use the notebook - rather the notebook forces you to use the API. The code can easily be moved into a stand-alone Python file for larger captures, more customization, etc.

The rest of the resource issues are.... known (lack of community support and related issues fall down to limited resources on our side, which I'm hopeful we can fix but it's been slow getting there).

colinoflynn, 2018-08-28:
(ps: @x8-999-github if possible to send a note to me at coflynn - AT - newae.com I had a quick follow-up question easier to take off-thread, but not sure of your contact as using the fancy github private emails)

alex-dewar, 2018-10-17:
I've been writing all the Jupyter stuff using python2 (my python3 installation is a little broken), but it's good to see that python3 support doesn't seem to be too far off. 

@adamws do you mind creating a pull request so I can merge this in?

alex-dewar, 2018-10-18:
B1 and B2 should now work for Python 3 as well.

There's a few syntax errors in Analyzer files when running setup.py, but they don't seem affect the installation.

h0rac, 2019-06-17:
Hey I found that API is using format strings that is supported only from python 3.6 >

serial_numbers.append(f"sn = {str(d['sn'])} ({str(d['product'])})")

```
  File "/usr/local/lib/python3.5/dist-packages/chipwhisperer-5.0.0-py3.5.egg/chipwhisperer/capture/scopes/openadc_interface/naeusbchip.py", line 83
    serial_numbers.append(f"sn = {str(d['sn'])} ({str(d['product'])})")

Invalid syntax
```

To fix this errors it require to write code in support to python 3 older than 3.6 like

`serial_numbers.append(f"sn = {sn} ({product})".format(str(d['sn']),str(d['product']) )`

Ubuntu 16.04 do not have python 3.6 for example

keesj, 2019-06-18:
Hi horac, are aware how old 16.04 is? why not upgrade to 18.04.2 LTS?

rob0tsunny, 2019-06-18:
check the VM image. we run inside virtual env to decouple from your local distributive. 
use this command to create it locally: pyenv virtualenv 3.6.7

closed 2021-03-25 by alex-dewar