newaetech/chipwhisperer #185: CWCapture reports wrong frequency for Nano

alex-dewar, 2018-11-07:
After setting the output clock frequency or ADC clock frequency for the CWNano, the reported value is incorrect. For example, setting an output frequency of 7.5E6 reports 7680000000 (7.68E9). The actual output clock is still correct.

Setting either value to `None` turns off the clock and reports the clock as 0, as expected. The actual clock value seems to be inversely proportional to the reported one (doubling the clock halves the reported value).

alex-dewar, 2018-11-29:
Probably the cause of the issue: https://github.com/newaetech/chipwhisperer/blob/1d8847e98968efab8d7448addbf1b581d38e78ba/software/chipwhisperer/capture/scopes/cwnano.py#L150

240E6 should probably be divided by bestdiv, not multiplied.

Same here:
https://github.com/newaetech/chipwhisperer/blob/1d8847e98968efab8d7448addbf1b581d38e78ba/software/chipwhisperer/capture/scopes/cwnano.py#L122

here:
https://github.com/newaetech/chipwhisperer/blob/1d8847e98968efab8d7448addbf1b581d38e78ba/software/chipwhisperer/capture/scopes/cwnano.py#L414

and here:
https://github.com/newaetech/chipwhisperer/blob/1d8847e98968efab8d7448addbf1b581d38e78ba/software/chipwhisperer/capture/scopes/cwnano.py#L437

alex-dewar, 2018-11-29:
Note that best_div can be 0 if the clock is set to `None`, so will have to account for that as well.

colinoflynn, 2018-11-29:
Good catch, this would be an error! Might need a small stub that does the conversion, and could deal with the 0/None safely

alex-dewar, 2018-11-30:
Fixed by commit 3fa31419ca19cac05d84ed6792c09a9bc73da08a

closed 2018-11-30 by alex-dewar