newaetech/chipwhisperer #187: [cw5dev] avoid spaces in notebook filenames

doegox, 2018-11-14:
Current notebook pages have spaces in their name.
This prevents cross-referencing them in Markdown from a notebook to another one with the syntax `[another tutorial](Tutorial A0 (blah).ipynb)`.
Please replace spaces by `-` or `_`.

colinoflynn, 2018-11-14:
Good catch! Will rename... had planned on changing the naming scheme a bit too (since 'basic' vs 'advanced' didn't make sense all the time anyway). So would be a good time to do so.

adamws, 2018-11-14:
One additional note from my side. If notebooks are meant to replace tutorials, you should commit them with outputs computed. Then you could even use service like [nbviewer](https://nbviewer.jupyter.org/) to share them outside github like this [book](https://nbviewer.jupyter.org/github/unpingco/Python-for-Signal-Processing/tree/master/) for example.

alex-dewar, 2018-11-14:
Should be fixed by f360b421b44a28d61f546cc517579e3d4c9f7de5. Thanks for the suggestion regarding the outputs. I've moved that to a separate issue

closed 2018-11-14 by alex-dewar