newaetech/chipwhisperer #196: XMEGA programmer fails with timeout when programming CW308 XMEGA target

alex-dewar, 2018-12-17:
This issue occurs on both Windows and in the VM, but not on the develop branch. It also only occurs on the CW308 XMEGA Target board and not when programming the CW303 XMEGA.

The develop programmer seems to detect the timeout and work around it, so it might be a bit of logic that was in the GUI (maybe the `autoProgram` method?) that didn't get moved over.

alex-dewar, 2018-12-17:
Issue fixed by 43410bc69efdddfe90aae39d9abb17911bbd273d

steelfly33, 2018-12-17:
I got this stack trace when trying the above patch in the VM:
```
INFO:root:Erasing Chip
Traceback (most recent call last):
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 277, in txrx
    response = self.usbdev().ctrl_transfer(payload[0], payload[1], payload[2], payload[3], payload[4], timeout=self._timeout)
  File "/usr/local/lib/python3.6/dist-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
WARNING:root:Full chip erase timed out. Erasing app only instead
Traceback (most recent call last):
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 245, in erase
    xmega.erase(memtype)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 198, in erase
    self.eraseChip()
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 479, in eraseChip
    self._xmegaDoWrite(self.XPROG_CMD_ERASE, data=[self.XPROG_ERASE_CHIP, 0, 0, 0, 0])
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 502, in _xmegaDoWrite
    status = self._xmegaDoRead(cmd=0x0020, dlen=3)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 511, in _xmegaDoRead
    return self._usb.readCtrl(self.CMD_XMEGA_PROGRAM, cmd, dlen)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 622, in readCtrl
    return self.usbseralizer.readCtrl(cmd, value, dlen)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 183, in readCtrl
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 149, in process_rx
    raise payload
usb.core.USBError: [Errno None] 110

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 277, in txrx
    response = self.usbdev().ctrl_transfer(payload[0], payload[1], payload[2], payload[3], payload[4], timeout=self._timeout)
  File "/usr/local/lib/python3.6/dist-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
Traceback (most recent call last):
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 245, in erase
    xmega.erase(memtype)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 198, in erase
    self.eraseChip()
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 479, in eraseChip
    self._xmegaDoWrite(self.XPROG_CMD_ERASE, data=[self.XPROG_ERASE_CHIP, 0, 0, 0, 0])
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 502, in _xmegaDoWrite
    status = self._xmegaDoRead(cmd=0x0020, dlen=3)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 511, in _xmegaDoRead
    return self._usb.readCtrl(self.CMD_XMEGA_PROGRAM, cmd, dlen)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 622, in readCtrl
    return self.usbseralizer.readCtrl(cmd, value, dlen)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 183, in readCtrl
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 149, in process_rx
    raise payload
usb.core.USBError: [Errno None] 110

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 70, in func_wrapper
    val = func(self, *args, **kwargs)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 248, in erase
[... 69 lines cut ...]
usb.core.USBError: [Errno None] 110

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./loader.py", line 33, in <module>
    programmer.erase()
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 73, in func_wrapper
    if self.scope.io.pdic != pin_setup['pdic']:
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/capture/scopes/cwhardware/ChipWhispererExtra.py", line 316, in pdic
    return self._getGpio(102)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/capture/scopes/cwhardware/ChipWhispererExtra.py", line 346, in _getGpio
    state = self.cwe.getGPIOState(pinnum)
[... cut ...]
    data = bytearray(self.serial.cmdReadMem(address, datalen))
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 630, in cmdReadMem
    return self.usbseralizer.cmdReadMem(addr, dlen)
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 196, in cmdReadMem
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 149, in process_rx
    raise payload
usb.core.USBError: [Errno None] 110
```

alex-dewar, 2018-12-17:
Replacing the `cw.programTarget(scope, prog, fw_path)` with:
```python
prog = prog()
prog.scope = scope
prog.open()
prog.find()
prog.erase(memtype="app")
prog.program(fw_path, memtype="flash", verify=True)
prog.close()
```

should fix things. 

It looks like, only in the VM, trying the full chip erase puts something into an unrecoverable state.

steelfly33, 2018-12-17:
That works.  Thanks @alex-dewar

steelfly33, 2018-12-17:
Traceback is still printed, but the code executes successfully with the warning.
```
INFO:root:Detected XMEGA128D4
INFO:root:Erasing Chip
Traceback (most recent call last):
  File "/home/vagrant/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 277, in txrx
    response = self.usbdev().ctrl_transfer(payload[0], payload[1], payload[2], payload[3], payload[4], timeout=self._timeout)
  File "/usr/local/lib/python3.6/dist-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
WARNING:root:Full chip erase timed out. Reinitializing programmer and erasing only application memory
INFO:root:Detected XMEGA128D4
XMEGA Programming flash...
XMEGA Reading flash...
```

alex-dewar, 2018-12-17:
Yeah, I was running into the traceback as well in the VM, but wanted to make sure the actual programming worked. I'll try to find out why the traceback is still being printed.

alex-dewar, 2018-12-18:
Doesn't appear to be a python 3.6 vs 3.7 issue either

annleeh, 2019-01-07:
Using ubuntu I also got the same issue. Replacing with the code above prog.find() also throws 

Traceback (most recent call last):
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 277, in txrx
    response = self.usbdev().ctrl_transfer(payload[0], payload[1], payload[2], payload[3], payload[4], timeout=self._timeout)
  File "/usr/local/lib/python3.7/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python3.7/site-packages/usb/backend/libusb0.py", line 593, in ctrl_transfer
    timeout
  File "/usr/local/lib/python3.7/site-packages/usb/backend/libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'error sending control message: Connection timed out'
Traceback (most recent call last):
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 70, in func_wrapper
    val = func(self, *args, **kwargs)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 232, in find
    sig, chip = xmega.find()
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 175, in find
    self.enablePDI(True)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 309, in enablePDI
    self._xmegaDoWrite(self.XPROG_CMD_ENTER_PROGMODE)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 502, in _xmegaDoWrite
    status = self._xmegaDoRead(cmd=0x0020, dlen=3)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 511, in _xmegaDoRead
    return self._usb.readCtrl(self.CMD_XMEGA_PROGRAM, cmd, dlen)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 613, in readCtrl
    return self.usbseralizer.readCtrl(cmd, value, dlen)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 183, in readCtrl
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 149, in process_rx
    raise payload
usb.core.USBError: [Errno None] None

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 285, in txrx
    response = self.cmdReadMem(addr, dlen)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 445, in cmdReadMem
    self.sendCtrl(cmd, data=pload)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 420, in sendCtrl
    return self.usbdev().ctrl_transfer(0x41, cmd, value, 0, data, timeout=self._timeout)
  File "/usr/local/lib/python3.7/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python3.7/site-packages/usb/backend/libusb0.py", line 593, in ctrl_transfer
    timeout
  File "/usr/local/lib/python3.7/site-packages/usb/backend/libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'error sending control message: Broken pipe'
Traceback (most recent call last):
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 70, in func_wrapper
    val = func(self, *args, **kwargs)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 232, in find
    sig, chip = xmega.find()
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 175, in find
    self.enablePDI(True)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 309, in enablePDI
    self._xmegaDoWrite(self.XPROG_CMD_ENTER_PROGMODE)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 502, in _xmegaDoWrite
    status = self._xmegaDoRead(cmd=0x0020, dlen=3)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/programmer_xmega.py", line 511, in _xmegaDoRead
    return self._usb.readCtrl(self.CMD_XMEGA_PROGRAM, cmd, dlen)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 613, in readCtrl
    return self.usbseralizer.readCtrl(cmd, value, dlen)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 183, in readCtrl
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 149, in process_rx
    raise payload
usb.core.USBError: [Errno None] None

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/capture/api/programmers.py", line 73, in func_wrapper
    if self.scope.io.pdic != pin_setup['pdic']:
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/capture/scopes/cwhardware/ChipWhispererExtra.py", line 316, in pdic
    return self._getGpio(102)
[... cut ...]
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 621, in cmdReadMem
    return self.usbseralizer.cmdReadMem(addr, dlen)
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 196, in cmdReadMem
    return self.process_rx(self.txrx(tx=cmdpacket))
  File "/home/catopsys/SCA/chipwhisperer/software/chipwhisperer/hardware/naeusb/naeusb.py", line 149, in process_rx
    raise payload
usb.core.USBError: [Errno None] None

alex-dewar, 2019-07-15:
The issues @annleeh was running into should be fixed by da1fd6ed76a9a2fbc4a982d2d7e22dc884d4512b and the warning from @steelfly33's last post has been changed to `logging.info` by 6ca266555ddae5779fb0c24fd1f14c23c67e7134.

closed 2019-07-22 by alex-dewar

muttiopenbts, 2019-08-07:
I'm running Ubunto 18
I just checked out develop branch and ran /chipwhisperer/jupyter/PA_Intro_1-Firmware_Build_Setup.ipynb script with USB timeout issues.
Made the changes suggested above to replace programTarget() with program() and ran perfectly.