newaetech/chipwhisperer #201: Unauthenticated Jupyter

floyd-fuh, 2019-01-04:
I used the new CW5 and I really enjoyed using the Jupyter notebooks. Thanks for going this hard route of changing all the tutorials, I hope it will pay out in the end as you won't have to maintain the GUI anymore.

However, I noticed that the Jupyter instance is not password protected. Additionally, the VM user has a static username and password (admittedly the SSH server seems to be properly configured). ChipWhisperer is a security product, so from that point of view I think having an unauthenticated Remote Command Execution (RCE) by leaving unauthenticated Jupyter access open is not a very good idea. It violates the secure-by-default principle.

I know there are reasons why you didn't implement authentication. It's probably something between "This is only for testing and in laboratory networks", "we properly state that the VM should have only a host interface", "a user should never expose the VM to the Internet", "we just started working on CW5 and it's beta" and maybe even a quite funny "but authentication is not user friendly". So I see your point. However, in 2018, the year of the Internet of Sh^WThings (where "but the user has to firewall these things" is a common hardware vendor excuse) and Shodan, I don't think this is a good enough. If you do a quick Shodan search you will see yourself that you can get RCE on several hundred Jupyter machines on the Internet. I don't think ChipWhisperer VMs should join that club.

The sane choice would either be to set a random long password on first startup or force the user to change the password before she can use it. I know this is not as easy as it sounds, but maybe you find a way to print the randomly generated password to the tty of the VM on first startup only or such. Or at least force the user to set a password before he can use the VM, this would still leave users vulnerable who just start the VM but never use it, but that's already a lot better. Just make it secure-by-default, so it's not definitely not your fault if one of the VMs pops up on the Internet.

colinoflynn, 2019-01-04:
For Windows we have/will have an installer that can avoid these issues by running everything as a normal program, which will likely be the recommended way for most users that aren't familiar with VMs (i.e., especially might not be aware that if the VM network is bridged to your local network it can be very bad).

The CW5 VM is (roughly) assumed to be running as a product install - i.e., you are using it on your local machine and thus this isn't a problem. Similar to many other VMs we defaulted to hard-coded passwords for now.

But I'm almost 100% certain someone will put one of these exposed at some point, so I'm leaving this open so we do indeed fix the problem (or at least force the user to acknowledge this). Having the password expired by default should be pretty easy I think at least for the VM.

Doing this on the Jupyter notebook might be more difficult. I'm going to leave this open and hope someone might know off-hand the best way forward, or for us to follow up once have more time to explore it! In the interm we could at least add filtering on the IP to ensure it's local requests only which is probably the quickest solution during the beta.

alex-dewar, 2019-05-16:
The latest VM release now requires the user to update the vagrant password as well as add a jupyter one.

closed 2019-05-16 by alex-dewar