newaetech/chipwhisperer #202: Virtual Box 6.0.0 doesn't work with CW5.Jupyter.20181124 image

steelfly33, 2019-01-09:
The VM does not boot.  Progress varies depending on serial port (not enabled boots further).  Downgrade to VB 5.2.2 and no issues.  Did not investigate further, but reproduced on multiple Windows 10 (host) systems.

rob0tsunny, 2019-02-09:
I was able to create a new image using stable debian and cw worked fine on MacOS. I think it has something to do with the kernel which is included to the VM. Serial port adding/removing changes the behavior, but doesn't really help.

alex-dewar, 2019-05-06:
https://github.com/hashicorp/vagrant/issues/10578

Looks like this is specific to running Ubuntu boxes with UART disabled. I've definitely run into issues booting with UART enabled before (though I strangely didn't run into them this time), so it may be best to just switch boxes. Both Debian and bento/ubuntu work.

alex-dewar, 2019-05-16:
Should be fixed as of the latest VM release

closed 2019-05-16 by alex-dewar