newaetech/chipwhisperer #204: Connecting CWCR2 setup_cw_generic.py

isep1302, 2019-02-15:
I know that the hardware is obsolete but I want to make it work, all my work has been done with my laptop (Linux-Ubuntu 18.04).

I'm stuck when I'm running the "setup_cw_generic.py" from CWCapture but I'm having the next error message:

   ```
 execfile('/home/tersites/chipwhisperer/software/chipwhisperer/capture/scripts/setup_cw_generic.py')
    Traceback (most recent call last):
      File "<console>", line 1, in <module>
      File "/home/tersites/chipwhisperer/software/chipwhisperer/capture/scripts/setup_cw_generic.py", line 12, in <module>
        scope.gain.gain = 45
    AttributeError: 'OpenADC' object has no attribute 'gain'
```
I really appreciate if you can help me to solve my problem with the setup.

PS: In case is helpful I have:

    ChipWhisperer Capture Complete Kit
    PN: NAE-CWCR2-KIT-COMP        REV:1.3

Regards,

colinoflynn, 2019-02-15:
The ChipWhisperer support for CWCR2 was removed in recent versions (4) - it sounds like you are using such a release? Basically the back-end is rather different, and the porting effort wasn't completed for that...

isep1302, 2019-02-15:
Thanks Colin for your answer. I'm using the current version (4). What's the last version that support CWCR2? 

Regards

alex-dewar, 2021-03-25:
Closing this as there's been no activity for a while.

I believe the last version to support the rev2 is 0.12RC1 (https://github.com/newaetech/chipwhisperer/releases/tag/0.12RC1) as it's marked as deprecated even in v3.

closed 2021-03-25 by alex-dewar