newaetech/chipwhisperer #207: cw-lite PCB specifications for USB High Speed

NeoFromMatrix, 2019-03-20:
There are rough specifications for the chipwhisperer lite PCB [/chipwhisperer-lite/pcb/gerbers/README.txt](https://github.com/newaetech/chipwhisperer/blob/develop/hardware/capture/chipwhisperer-lite/pcb/gerbers/README.txt)

But I could not find anything regarding the impedance control of the PCB. Afaik USB HS can be picky and choosing the wrong PCB material/no impedance control might prevent it from working properly. 

Has anyone experience with ordering PCBs for this design and which impedance controlled material to use?

colinoflynn, 2019-03-20:
USB HS is pretty unpicky in reality - I'd just use "whatever" and not worry about it. The mismatch segment is relatively small compared to the rest of the cable etc.

closed 2019-03-20 by colinoflynn

NeoFromMatrix, 2019-03-20:
Ok. Thanks a lot for the fast answer!