newaetech/chipwhisperer #208: Chipwhisperer bricked?

tresacton, 2019-03-27:
I launched CW Capture and pressed the button to connect the device and scope. In the console it says they connected, the version numbers don't get printed, and the buttons in the GUI for connect remain red. If I continue to press connect, the same behaviour repeats with the buttons never turning green.

I decided that it might be bricked, and that I should probably install Bossa and attempt to flash the device. However, the chipwhisperer doesn't show up in the comms port list. I also don't have an option to flash the firmware in the Tools menu in Chipwhisperer (I currently have the most recent binary installed which is v4.0.1.

I opened device manager, made sure the chipwhisperer is pointing at the drivers folder in its program folder and reinstalled the device.

The chipwhisperer was previously functional. The only thing that occurred was that my computer went to sleep when it was connected (but not currently performing a trace), and the problems started happening when I woke the computer up. I've tried rebooting the computer, and reinstalling the chipwhisperer software.

I'm not sure what to do at this stage.

When powered on, the chipwhisperer pulses a blue led.

tresacton, 2019-03-27:
Edit: I am able to reprogram the device using Bossa. That one was definitely user error, it worked as soon as I shorted the erase pins!

I still have the problems I described above after flashing the firmware (C:\Program Files (x86)\ChipWhisperer\chipwhisperer\hardware\capture\chipwhisperer-lite\sam3u_fw\SAM3U_VendorExample\Debug\SAM3U_CW1173.bin).

I flashed the firmware, ensuring all options/checkboxes matched those in the wiki screenshot. I closed down Bossa, powercycled the chipwhisperer (unplugged USB, and plugged back in), confirmed the driver was as it was before I programmed it (i.e. showing up lib-usb32-devices as ChipWhisperer Lite), launched CWCapture, and attempted to use connect buttons. Same result as before I programmed it - the buttons stay red, the console reports it was connected, there are no version numbers, the scope and target tabs are empty and the device is not usable. :'(

colinoflynn, 2019-04-09:
Sorry for missing this issue - the CW should not just "die" as such, so it's either a rare hardware failure or some other driver problem. Is there any specific error reported? Have you tried on other machines by chance or on the Virtual Box edition?

tresacton, 2019-04-25:
Hi Colin, thanks for the response.
I haven't had a chance to check the virtual box edition, but can confirm the behaviour is the same on another machine with a fresh, first-time, install of the software (pre-Jupyter versions, I haven't looked into the new software as of yet).

No driver modifications, or the installation of any related/unrelated software, were performed between everything being functional and the odd behaviour described, and the machine is regularly backed up. Reverting to a backup before the issue started occurring didn't resolve the problem.

fluquin8, 2021-05-22:
Hi tresacton,

were you able to figure out a solution? I also left the chip whisperer connected when my computer went to sleep, and now I have the same issue.

colinoflynn, 2021-05-22:
Huh - does it come up as anything on the USB? This does sound like an odd problem. What is your host (windows?).

Does it show up as a USB-Serial port by chance? If that is the case it sounds like it entered bootloader mode. There is a lot more needed on the USB side to do that so just going to sleep shouldn't be an issue. But if it's in bootloader mode you can just reload the firmware with the update too.

The "faint red & blue lights" does sound like bootloader mode, which is what makes me suspect that.

fluquin8, 2021-05-22:
Hi Colin,

Thank you for the quick response! I am relatively new to CW, so I will try to answer questions to the best of my ability.

I am using CW specifically for the Manual CPA attack, so I have only used it through (and need it for) the PA_CPA_2-Manual_CPA_Attack Jupyter Notebook. I think the version for this is 5.3.0 or 5.3.1.

Before, the CW ran smoothly, but now when I run '1.2.1 Setup' section, or specifically '%run "Helper_Scripts/Setup_Generic.ipynb"', I get 'Warning: Failed to find USB Device." I have modified some code in the AES encryption, but just strictly the AES -- nothing that would affect the firmware. That is why tresacton's post caught my attention; I remembered I left the CW plugged in for a while before coming back to it. 

The CW's is owned by a lab in my university, so I am cautious about firmware updating and making the situation worse (bricking it).

Can you point me to how I can reload the firmware?

Thank you!

colinoflynn, 2021-05-22:
Hi fluquin8,

So there is a "rom" bootloader on the device, in theory it's unbrickable! The firmware update instructions are at https://rtfm.newae.com/Capture/ChipWhisperer-Lite/#upgrading-sam3u-firmware . You might try to short the "erase" jumper to ensure it *is* in bootloader mode:

But if you are on Windows it's easier to use the stand-alone "firmware update tool". The latest version can be downloaded from here: https://github.com/newaetech/chipwhisperer/releases/tag/5.5

If it's in firmware update mode, it should show up in device manager as "AT91 USB to Serial Converter"
[image]

colinoflynn, 2021-05-22:
PS - if it works it should look like this:
[image]

fluquin8, 2021-05-22:
I shorted the erase jumper, and the blue and red LEDs remain dimly list. However, I am not seeing anything show up in my Ports (COM & LPT) tab. (Well, the tab doesn't show up because nothing is connected.)

Do you have any suggestions or thoughts on this?

colinoflynn, 2021-05-22:
Does it show up as a COM port on any other computers (or - it might show up as needing a driver, I think the bootloader serial port requires a driver). But you should hear the windows 'device detected' sound. It would work on any computers (Win/Linux/Mac) to at least detect that.

I forget what it's called when there is no driver installed - but it would show up as something with the yellow exclamation mark if it's detected but needs the driver. The SAM-BA driver is the standard Atmel 6124.inf driver (sorry I'm not at office so don't have as good refs - but anything like https://startingelectronics.org/articles/atmel-ARM/SAM-BA-USB-CDC-driver-install/).

If it's not showing up at all, and no windows "device attached" sound, something could be wrong with the HW. Double-check if you unplug-replug theUSB on the CW-Lite side it doesn't change. Otherwise we can swap the HW (drop a note to support@newae.com) but that will obviously be slower than if we can figure out a reason why it stopped working!

fluquin8, 2021-05-22:
No, it does not show up as a COM port for any other computers. I tried it on another Windows and Mac computer. I don't get any alerts or sounds. Unfortunately, I am not familiar enough with any OS to troubleshoot this as in depth as I would like to.

Of the top of your mind, do you know if there are any other ways to troubleshoot why it is not appearing or how to get it to appear? I am using the CW-Lite for a final project, and I think a replacement would arrive to late. But if nothing can be done, nothing can be done.

Specifically for my final project my partner and I are building out a side channel attack engine on an FPGA that performs correlation power analysis. We just need the target board to perform AES so we can measure from the SMA port. Do you think it would be doable to configure the target with another microcontroller (like STM32F4) using code already on CW's github? Any general starting point or tidbits would be appreciated.

Thank you Colin! I really appreciate your responsiveness :)

colinoflynn, 2021-05-22:
Hmm - really unsure what happened there! It sounds like the main micro isn't coming up then. If you had a JTAG device you could mount the 20-pin header and see if that shows the core being alive, but to be honest even if the micro is alive & USB is dead it's still not great. The main bootloader is fully in ROM, so if it's not working normally it means other issues anyway.

There isn't much USB debugging needed beyond what you described - it should mostly just work, especially once it's forced into bootloader mode by shorting the "erase" jumper.

With a DMM you could check the voltages is a quick debug step. The schematic at https://github.com/newaetech/chipwhisperer/blob/develop/hardware/capture/chipwhisperer-lite/pcb/cw-lite-main.pdf but in particular, you should see 1.8V on the VDDCORE rails, any of the caps here should have that:

[image]

FYI - there is a holiday here on Monday so no shipping, but we could send a replacement Tuesday. If you're in north america it is normally 1-2 days, EU 2-3 days. So not sure how quickly it's needed, but you may yet get something this week if helpful.

colinoflynn, 2021-05-22:
PS - are you just using the target side mostly then? If so yes you could compile the code on another STM32Fx target, it should kinda "just work". We don't use any peripherals really besides USART & clock. The default code assumes a 7.37MHz clock coming in is the only real requirement.

fluquin8, 2021-05-22:
Hi Colin,

I just spoke with my instructor, and I will send an email to support@newae.com for the swap. We are in the last two weeks of the quarter, so we are trying to minimize our risk/stress. We are in the U.S., and we would pay for the fastest/most reasonable shipping fee, so I think this would be fine.

Should I reference this discussion?

colinoflynn, 2021-05-23:
Sounds good! Drop a note there - we'll need shipping address + phone # (for courier only) - you can xref this in case I don't get pinged right away, we only get a few warranty issues a month still so most of the time it's pretty easy. But I'll make sure it doesn't get dropped as know about the speed issue, we'll send new one first & figure out return after (I'm curious to get this back as it might be a HW failure which are fairly rare, and would like to track what happened).

fluquin8, 2021-05-23:
Okay, I sent the email.

Sounds good. I'll be on the look out for the return details.

Once again, thank you!!

fluquin8, 2021-05-30:
Hi Colin,

I received the CW capture board. Thank you!

I am not sure if this issue tab is the best place to ask, but I need some help setting up the CW. I tried to run the jupyter notebook as I would previously, and I get the error 'NameError: name 'scope' is not defined' and 'usb.core.USBError: [Errno None] b'could not set config 0: Device or resource busy'.

I also tried to run this code I saw on the readthedocs guide: 
`import chipwhisperer as cw`
`scope = cw.scope()`
`scope`
This gives me 'Could not connect to "NewAE USB (CWLite/CW1200)". It may have been disconnected, is in an error state, or is being used by another tool.' However, VM VirtualBox recognizes the CW, but the CW does not show up in the device manager.

When I disconnect the CW from the VirtualBox in Devices>usb tab so that the CW shows up in device manager and run the code again, I get 'OSError: USB Device not found. Did you connect it first?'

I am not sure what to do.

colinoflynn, 2021-05-31:
Hello,

We did some updates to the ChipWhisperer firmware - you might be running into a problem with a newer version of the firmware not working with an older version of the software, I'm not sure of the exact compatibility.

If you unplug & replug the ChipWhisperer, and run this code:
```
import chipwhisperer as cw
scope = cw.scope()
```

Does it work once? One change with the firmware is you need to disconnect from it before reconnecting. Previously it would automatically "override" and steal the connection I think, so if you accidently had a double-connect in the script somewhere, or if you connect in one tab and move, it won't work anymore.

If that doesn't work, can downgrade the ChipWhisperer firmware OR upgrade the VM. A real quick thing would be to download the latest release VM and just confirm it works in that (https://github.com/newaetech/chipwhisperer/releases/tag/5.5.2).

fluquin8, 2021-05-31:
It worked with the version you sent me! I was able to make the minor changes to use the archived tutorial/journal I needed, and now everything is set up the way I would like it to be. Thank you for your patience and quick responses. (I will send the other CW tomorrow. Today is a holiday here.)

rtborg, 2021-09-27:
Hi,

I am faced with the same problem - CW Lite will not connect after left connected to a Windows PC, which went into a sleep mode. Now the CW blinks the blue LED (USB Enumerated), and shows as a USB Serial Device under the Device Manager Ports category.

I tried using the Virtual Box installation, the CW is shown as an Unknown Device in the USB Devices list:
[image]

When I run the `Setup_Generic` script, I get the following message:
`
OSError: Unable to communicate with found ChipWhisperer. Check that                 
another process isn't connected to it and that you have permission to communicate with it.`

citizen6874, 2022-04-10:
@rtborg 
I have a connection issue with my new CWLite and Windows10.  I tried your approach (Clear SAM firmware, reflash).  I went from flashing blue LED (1 Hz) to very dim red+blue leds after shorting the connection.  When I follow your reflash command, though, I get the following:

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_74468/565863851.py in <module>
      1 import chipwhisperer as cw
----> 2 cw.program_sam_firmware(hardware_type='cwlite')

AttributeError: module 'chipwhisperer' has no attribute 'program_sam_firmware'

I don't understand why I would get this, and am stuck.  Have you got any suggestions?
Thanks,
pro731

lo3s, 2022-11-07:
Hi,

I have similar issue than previously discussed.
When I run:

> import chipwhisperer as cw
> scope = cw.scope()

I have the following error:

> **OSError: USB Device not found. Did you connect it first?**

I tried several things already:
reinstalling the drivers from windows using **zadig-2.7**.
adding the **libisb dll**.
erasing the frimware from the cwlite and programming it again using **bossa-x64-1.9.1**.
erasing the frimware from the cwlite and programming it again using **cw_firmware_updater_5_5_0**.

I can get the blue led to blink again and windows to recognize cw lite in the list of devices. However I still cannot cuse cw.scope() successfully.

Any leads ?

Cheers,

alex-dewar, 2022-11-07:
@citizen6874 You've got an old version of ChipWhisperer. You either need to update to 5.6.1, or use the older programming method: https://chipwhisperer.readthedocs.io/en/latest/scope-api.html#api-scope-update

alex-dewar, 2023-10-13:
Closing this issue, as it's pretty off topic and it's difficult to give support through old issues like this. For support, contact support@newae.com or make a thread on https://forum.newae.com/

closed 2023-10-13 by alex-dewar