newaetech/chipwhisperer #209: getting started issues sca level 2 kit

jconway1970, 2019-04-08:
I am still having trouble getting started with your SCA Level 2 Kit stepping through your examples.  Here is some feedback.
On Windows10 x64 I had trouble with the VMWare video and usb drivers.  After fiddling and giving up on them, I had more luck with the Manual Install method.  (I am able to hand trigger a single Capture). 
I get an error with WinAVR and Studio 7 running make. (rm.exe Application Error 0xc0000142,  “Error 128”), but a hex file is still produced.
If I try to flash the hex code that is produced using an Atmel Ice to the target ATxmeaga 128D4, I get Read voltage 0.7V is outside selected device’s operating range: 1.6V to 3.6V.  

Is there a jumper/software setting I need to make to make the flash work? 
Your examples show running Example Scripts, but this is removed.  Is there updated tutorial information that shows how to run examples with Capture V4?

jconway1970, 2019-04-08:
Also, do you have any documentation/example with P/N NAE-CW308-04 red UFO Target board?

jconway1970, 2019-04-09:
I figured it out.  WinAVR doesn't seem to work with the makefile and windows 10.  I removed and replaced with Atmel Studio 7, and was able to open a project from the make file.  I was able to create the flash elf file and load on the chip.  I have successfully run the serial AES attach.

colinoflynn, 2019-04-09:
Sorry was trying to hunt that down - this is a new bug so wasn't something I had an immediate fix for! There is no jumper etc so it "should" have worked. For the other questions:

1) What version of the SW are you using, V4? The 'example scripts' should still be there.
2) The UFO documentation is mostly on the wiki right now (http://wiki.newae.com/CW308_UFO_Target), am looking at updating that with a new format to better show all the features however.

jconway1970, 2019-04-09:
Yes, V4.0.2.  I think the issue is with winavr and windows 10.  I was able
to compile in Atmel Studio 7.  I spent some time investigating the error,
to no avail.

colinoflynn, 2019-04-09:
For Windows 10 - this might be covered by the 8.1 fix. There is a small non-obvious note on http://wiki.newae.com/Installing_ChipWhisperer which says "Note if using WinAVR on Windows 8.1, you must replace the dll msys-1.0.dll with an updated version.", which has a link to this: https://www.avrfreaks.net/forum/windows-81-compilation-error

The proper fix is to remove WinAVR from the docs/suggested tools I think, as it's a very outdated toolchain at this point! It was the easiest method to install for a while, but now with the Windows 10 errors it may no longer be worthwhile...

alex-dewar, 2021-03-23:
Closing, as we no longer recommend installing WinAVR

closed 2021-03-23 by alex-dewar