newaetech/chipwhisperer #213: Quite confused with new interface

nico0481, 2019-05-20:
Hi,

I'm quite confused with new interface of CW (Jupyter), I have installed VM, ChipWhisperer 5.0.1alpha, I can view Jupyter Notebook... but how to start CW GUI??
Device seems to be recognize because drivers are now installed (Windows) but I have no idea how to work with.

Regards

Nico

alex-dewar, 2019-05-20:
Hi Nico,

The GUI has been replaced by the Notebooks for CW5. You can find updated tutorials for CW5 in the software/jupyter directory.

Let me know if you need more information.

Alex

nico0481, 2019-05-21:
Hi Alex,

I don't easilly understand how it works, sorry.
Should I "run" every code exerpts? 
I.e. if I try to run "PA_Intro_1-Firmware_Build_Setup " , I get error in point 4 (Python Script): 
"Warning: Failed to find USB Device"
Device appears in Windows Device Manager as lib-usb win32 device, Chipwhisperer CW1200.
On device itself, it shows "FPGA Unprogrammed-Waiting on Bitstream"...
I went trough each Jupyter items but don't find solution.

Thank you
Regards
Nico

alex-dewar, 2019-05-21:
Hi Nico,

For the most part, you should be running all the code blocks, though you may need to change some code up for different hardware. 

Regarding your USB issue, I think I forgot to add a filter for the Pro. You can add a new filter by shutting down the VM, selecting the VM in Virtualbox, and going to Settings/USB. On this screen, there should be a little USB+ icon that you can use to select the Pro. Once that's done, start the VM again and you should be able to run the Jupyter code blocks.

Let me know if that solves your issue.

Alex

colinoflynn, 2019-05-21:
As another short note - CW5 has been in "beta" status much longer than I planned... I think one very useful thing here would be some improved getting started documentation including an overview video. This has been on an internal TODO list of mine as part of the final release, but I hadn't had time to talk with others here about making this happen in real life. Your comment about it being unclear made it very much a requirement I think for us before final release.

lordoftheseas, 2019-05-22:
Hi,

Could someone please list out the steps I need to follow to get this working on a Mac? Also, do I install the files in the release in the virtual box or in the C drive?

AsFaBw, 2019-06-19:
I bought a CW-lite and started yesterday with installation and exercising the first lessons.
Basically you already recognized, but I just would like to give you a positive feedback on your plans and share my small beginner issues:
 *using "pip" and "python" in installation process caused issues as having installed 2.7 in parallel to 3. So a "pip3" and "python3" solved that beginner issue for me.  
 *I was not sure from where to start "jupyter notebook". If you have once used jupyter this is not a riddle at all, but for the first time usage without knowing what jupyter looks like or behaves a clear "cd chipwhisperer", click on "...","...." instruction would take uncertainties for the first entry.
 *The "!!Suggested_Completion_Order!!.ipynb" is a great solution for easy entry. Nevertheless my immediate feeling was "A jupyter introduction tutorial could have been in front of the Power Analysis tutorial" to understand what is exactly exercised in the background and why:-)

Compared to other riddles with other gadgets of similar class or even mass products, I feel that the documentation so far is great - well done! Some googling and having a look to CW independent jupyter tutorials can be expected from me - especially as long as CW5 if not officially released.

Thanks a lot so far!

AsFaBw, 2019-07-25:
I have seen the new jupyter notbook tutorial. I think that really helps. Thanks a lot!

In between I came across the jupyter documentation, that could be a further recommended read at the end of your introduction tutorial. Below is the link:

https://jupyter-notebook.readthedocs.io/en/stable/

alex-dewar, 2019-07-26:
Thanks for the suggestion! I think we're planning on doing a 5.1.2 release fairly soon to smooth things out a bit, so I'll make sure we include that link for that release.

FranzHeubach, 2019-07-30:
@AsFaBw We have added your suggestion to the **chipwhisperer** [ReadTheDocs documentation](https://chipwhisperer.readthedocs.io/en/latest/starting.html) and have introduced an Introduction To Jupyter notebook in the jupyter directory in the develop branch. This will be included in master branch in the next release 5.1.2.

alex-dewar, 2022-01-27:
Closing as there's been no activity for a while

closed 2022-01-27 by alex-dewar