newaetech/chipwhisperer #216: CW5/CW1200 serial recieve loss

Wmyers559, 2019-06-26:
I seem to be having some issues with Chipwhisperer 5 (git commit(s): 24d29540 and bca19c4f) with receiving serial communications from a Chipwhisperer Pro connected to a CW308 board, with both the XMEGA and STM32F3 targets. 

There seems to be some loss with any serial communication between the Chipwhisperer and the target, and I'm reasonably sure that it is occurring on the CW1200 end. I've probed the TX/RX lines with a logic analyzer, and the results are as expected, with the proper response from the DUT (this was verified on the `simpleserial-base` firmware, but a similar thing happens on just about all of the other notebooks that I have tried.).

For example, here is from a clean install with the default Jupyter notebook for PA_Intro1...:

[image]
[image]

Any suggestions?

Wmyers559, 2019-06-26:
Note: messing with the timeout for the `ser.read` call does not seem to fix the problem, and the number of nybbles lost does not seem to be constant--it varies between each run. It's also not necessarily on byte boundaries, as I have gotten an odd number of nybbles.

Wmyers559, 2019-06-26:
Update: _here_ it appears to be related to the `print(output[1:-5].encode("ascii"))` line, which cuts off the last couple characters when the response from the device is missing the `\nz000\n`? Will update with the results of looking at this for the Simple Timing Analysis notebook.

Wmyers559, 2019-06-26:
Part of the problem appears to be that the `print(output[1:-5].encode("ascii"))` here cuts off the last few characters when the end of the response is sooner than the chipwhisperer software expects, but that doesn't fully explain the issue. Here is another set of results, showing that the software is only getting a partial packet from the device:

(Edit: web proxy doesn't like me uploading images right now, will post in a moment.)

Wmyers559, 2019-06-26:
[image]

[image]

alex-dewar, 2019-06-26:
Hi,

This might be caused by doing the serial read too soon after capturing - i.e. the target might not have sent back all its data by the time you do the read. If you throw a `time.sleep(0.25)` in after `cw.capture()`, does the full string get printed back consistantly?

Wmyers559, 2019-06-26:
That seems to be the issue with the PA_Intro1! Looking at PA_SPA_1-Timing_Analysis..., I get a similar issue, but increasing/tweaking the delays does not seem to help. The exact output is:
```
WARNING:root:SAM3U Serial buffers OVERRUN - data loss has occurred.

*****Safe-o-matic 3000 Booting...
Aligning bits........[DONE]
Checking Cesium RNG..[DONE]
Masquerading flash...[DONE]
Decrypti PUNISHED
Please enter password to continue: 
```
Based on the comments [here](https://forum.newae.com/t/serial-buffer-issues/1242/3), the warning isn't really an issue, but trying the fix mentioned there (modifying the target code) doesn't seem to have helped. As before, the logic analyzer confirms that everything is passed from the target. (I can post images if necessary, but I don't want to clutter the thread too much)

Wmyers559, 2019-06-26:
Playing with the timing analysis one a bit more seems to reveal that the missing bytes always occur _after_ the `Decrypi` of the `Decrypting database..[DONE]` line, regardless of the delays in the target firmware.

Wmyers559, 2019-06-26:
From playing with delays in the block and in the target firmware, it looks like it reads 127 characters, which gets up to `...Decrypti`. Then it seems like there is already more data in the serial buffer when that read occurs, which then flushes that data, and then loop catches the last bits of data as they arrive. This is an issue with both the XMEGA and the STM32F3 targets (All that I have to test on right now.)

Is there anything that I could do to look into this further other than just continuing my black-box probing? I'd love to help contribute to this.

alex-dewar, 2019-06-26:
The reason this happens is that there's a limited buffer on the ChipWhisperer (the scope side, not the target side). Once this buffer is full, it stops inserting data, which is why you see reading stop after 127 characters.

Wmyers559, 2019-06-27:
Ah, cool. I've managed to mitigate by increasing the delay in the target firmware -- would you like me to put that together in a pull request for you?

Thanks for your time!

alex-dewar, 2019-06-27:
Thanks for looking into this. I should have a commit coming in a bit that increases the delay. I'm also going to add some additional explanation to the tutorials to better explain how the serial communication with the target works.

Wmyers559, 2019-06-27:
Cool, that would be great! Is there anything that you want help with? I've got a bit of spare time at the moment, and would love to give you a hand.

alex-dewar, 2019-06-27:
Well, we're currently going through a lot of restructuring and changes in the project in preparation for the release of CW5, but if you don't mind working through that and giving some feedback, we'd love the help! 

https://wiki.newae.com/V5:Installing_ChipWhisperer/Installing_ChipWhisperer_from_Git should be updated with instructions on how to work with the new structure of the repo (really the only big difference here is that the jupyter stuff is now in a submodule). Let us know if you run into issues with that new install proceedure.

closed 2019-07-01 by Wmyers559

Wmyers559, 2019-07-01:
I'm going through and mentioning any further issues that I encounter/fix in pull request #218, since any more would be somewhat off topic here.