newaetech/chipwhisperer #220: Update Pro only features for CW5

alex-dewar, 2019-07-02:


Wmyers559, 2019-07-02:
What would need to be done to take care of this? I'm willing to help if I am able to.

alex-dewar, 2019-07-02:
Currently, not much work has been done getting the pro specific features (SAD trigger, stream mode, and decode trigger) ready for CW5. The current plan is to get the interface cleaned up a bit, get it documented, then write some notebooks about how to use them.

If you do want to work on it, the decode trigger (cwhardware/ChipWhispererDecodeTrigger.py) is probably the best place to start, since that looks to be the least reliant on the GUI. Even just removing param stuff and verifying everything still works would be helpful.

alex-dewar, 2019-07-03:
The Pro's stream mode should now work. Any long capture won't display well using any browser based graphing utility, so this mode probably won't be much use until the Lite GUI gets implemented.

alex-dewar, 2019-07-04:
The Pro's SAD now works. The `scope.trigger.module` can now be used to set different modules if the Pro is connected, but remains readonly for the Lite. The SAD trigger module now appears when printing the scope object.

alex-dewar, 2019-07-04:
Decode trigger now works as well.

closed 2019-07-11 by alex-dewar