newaetech/chipwhisperer #229: CW308 STM32F Target Board FLASH error

chaeeunei, 2019-07-15:
Hello, I am Eun-Ei studying at Kookmin university in South Korea.

I had been bought ChipWhisperer Board.
(link : http://store.newae.com/cw308-ufo-board/) 

I am trying to execute as follows : Tools - Serial STM32F Programmer - Erase/Program/Verify FLASH.

I had made its 'makefile' well. I think that it doesn't matter.

But It doesn't well and have some problems with errors.

The error is as follows : 
An error has occurred:
Can't read port or timeout (Traceback (most recent call last): File "C:\Program Files (x86)\ChipWhisperer\chipwhisperer\software\chipwhisperer\hardware\naeusb\programmer_stm32fserial.py", line 282, in _wait_for_ask ask = self.sp.read(1)[0] IndexError: list index out of range )

It is usually safe to continue, but save your work just in case.
If it persists, try reseting the settings first before creating a new ticket informing the details bellow.

(show details is as follows : 
---------------------------
Error
---------------------------
An error has occurred:<br>Can't read port or timeout (Traceback (most recent call last):
  File "C:\Program Files (x86)\ChipWhisperer\chipwhisperer\software\chipwhisperer\hardware\naeusb\programmer_stm32fserial.py", line 282, in _wait_for_ask
    ask = self.sp.read(1)[0]
IndexError: list index out of range
)<br><br>It is usually safe to continue, but save your work just in case.<br>If it persists, try reseting the settings first before creating a <a href='https://github.com/newaetech/chipwhisperer/issues/'>new ticket</a> informing the details bellow.
---------------------------
Close   Hide Details...   
---------------------------
)

How solve the problem?

Please check and respond to me.

Best Regards,

Eun-Ei.

alex-dewar, 2019-07-15:
Hi,

There should be a "Read Block Size" option in the window for the STM32F Programmer. Can you try changing this from 256 to 64 and seeing if the problem persists?

Alex

chaeeunei, 2019-07-16:
Thanks your request.

Yes.
I try to execute your advice, but it still has some matter.

chaeeunei, 2019-07-16:
Thanks your request.

Yes.
I try to execute your advice, but it still has some matter.

Eun- Ei

ukuhne, 2019-07-16:
Hi, 
I seem to have the same (or a related) issue with STM32 target boards. I am using the CW308 with either a STM32F0 or F3 target board. I am under Linux and currently testing the cw5dev development branch for ChipWhisperer 5. In the Jupyter notebook, when trying to program one of the STM32 targets, I get this error:
```
Detected known STMF32: STM32F302xB(C)/303xB(C)
Extended erase (0x44), this can take ten seconds or more
Attempting to program 6019 bytes at 0x8000000
STM32F Programming flash...

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in _wait_for_ask(self, info)
    286         try:
--> 287             ask = self.sp.read(1)[0]
    288         except:

IndexError: list index out of range

During handling of the above exception, another exception occurred:

CmdException                              Traceback (most recent call last)
~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in writeMemory(self, addr, data, smallblocks)
    619             try:
--> 620                 self.cmdWriteMemory(addr, data[offs:offs + block_size])
    621             except CmdException:

~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in cmdWriteMemory(self, addr, data)
    419             self.sp.write(chr(crc))
--> 420             self._wait_for_ask("0x31 programming failed")
    421             logging.debug("    Write memory done")

~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in _wait_for_ask(self, info)
    288         except:
--> 289             raise CmdException("Can't read port or timeout (%s). Target didn't respond when an ack was expected." % traceback.format_exc())
    290         else:

CmdException: Can't read port or timeout (Traceback (most recent call last):
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 287, in _wait_for_ask
    ask = self.sp.read(1)[0]
IndexError: list index out of range
). Target didn't respond when an ack was expected.

During handling of the above exception, another exception occurred:

IndexError                                Traceback (most recent call last)
~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in _wait_for_ask(self, info)
    286         try:
--> 287             ask = self.sp.read(1)[0]
    288         except:

IndexError: list index out of range

During handling of the above exception, another exception occurred:

CmdException                              Traceback (most recent call last)
<ipython-input-33-b876eeece5c0> in <module>
----> 1 cw.program_target(scope, prog, fw_path)

~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/__init__.py in program_target(scope, prog_type, fw_path, **kwargs)
     52     prog.find()
     53     prog.erase()
---> 54     prog.program(fw_path, memtype="flash", verify=True)
     55     prog.close()
     56 

~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/capture/api/programmers.py in func_wrapper(self, *args, **kwargs)
     68             self.scope.io.nrst = 'high_z'
     69         try:
---> 70             val = func(self, *args, **kwargs)
     71         finally:
     72             logging.debug('Restoring pdic, pdid, and nrst pin configuration')
[... 47 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 620, in writeMemory
    self.cmdWriteMemory(addr, data[offs:offs + block_size])
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 420, in cmdWriteMemory
    self._wait_for_ask("0x31 programming failed")
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 289, in _wait_for_ask
    raise CmdException("Can't read port or timeout (%s). Target didn't respond when an ack was expected." % traceback.format_exc())
chipwhisperer.hardware.naeusb.programmer_stm32fserial.CmdException: Can't read port or timeout (Traceback (most recent call last):
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 287, in _wait_for_ask
    ask = self.sp.read(1)[0]
IndexError: list index out of range
). Target didn't respond when an ack was expected.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 287, in _wait_for_ask
    ask = self.sp.read(1)[0]
IndexError: list index out of range
). Target didn't respond when an ack was expected.
```
This is the first time I tried to use the ARM targets. Do I need to change any of the many switches on the CW308? I also tried putting a jumper between SH- and SH+ without success. Thanks for any hint how to proceed.

alex-dewar, 2019-07-16:
Hi,

Can you both try the latest commits for ChipWhisperer? I was able to recreate the issue and I believe 4ce454477254416a7568efb5e8b5e1693bc17d25 should fix it.

ukuhne, 2019-07-17:
This did not fix the issue for me. Same problem as before:
```
Detected known STMF32: STM32F071xx/STM32F072xx
Extended erase (0x44), this can take ten seconds or more
Attempting to program 4891 bytes at 0x8000000
STM32F Programming flash...

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in _wait_for_ask(self, info)
    285         try:
--> 286             ask = self.sp.read(1)[0]
    287         except:

IndexError: list index out of range

During handling of the above exception, another exception occurred:

CmdException                              Traceback (most recent call last)
~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in writeMemory(self, addr, data, smallblocks)
    618             try:
--> 619                 self.cmdWriteMemory(addr, data[offs:offs + block_size])
    620             except CmdException:

~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in cmdWriteMemory(self, addr, data)
    418             self.sp.write(chr(crc))
--> 419             self._wait_for_ask("0x31 programming failed")
    420             logging.debug("    Write memory done")

~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in _wait_for_ask(self, info)
    287         except:
--> 288             raise CmdException("Can't read port or timeout (%s). Target didn't respond when an ack was expected." % traceback.format_exc())
    289         else:

CmdException: Can't read port or timeout (Traceback (most recent call last):
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 286, in _wait_for_ask
    ask = self.sp.read(1)[0]
IndexError: list index out of range
). Target didn't respond when an ack was expected.

During handling of the above exception, another exception occurred:

IndexError                                Traceback (most recent call last)
~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py in _wait_for_ask(self, info)
    285         try:
--> 286             ask = self.sp.read(1)[0]
    287         except:

IndexError: list index out of range

During handling of the above exception, another exception occurred:

CmdException                              Traceback (most recent call last)
<ipython-input-14-b876eeece5c0> in <module>
----> 1 cw.program_target(scope, prog, fw_path)

~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/__init__.py in program_target(scope, prog_type, fw_path, **kwargs)
     52     prog.find()
     53     prog.erase()
---> 54     prog.program(fw_path, memtype="flash", verify=True)
     55     prog.close()
     56 

~/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/capture/api/programmers.py in func_wrapper(self, *args, **kwargs)
     68             self.scope.io.nrst = 'high_z'
     69         try:
---> 70             val = func(self, *args, **kwargs)
     71         finally:
     72             logging.debug('Restoring pdic, pdid, and nrst pin configuration')

[... 38 lines cut ...]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 619, in writeMemory
    self.cmdWriteMemory(addr, data[offs:offs + block_size])
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 419, in cmdWriteMemory
    self._wait_for_ask("0x31 programming failed")
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 288, in _wait_for_ask
    raise CmdException("Can't read port or timeout (%s). Target didn't respond when an ack was expected." % traceback.format_exc())
chipwhisperer.hardware.naeusb.programmer_stm32fserial.CmdException: Can't read port or timeout (Traceback (most recent call last):
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 286, in _wait_for_ask
    ask = self.sp.read(1)[0]
IndexError: list index out of range
). Target didn't respond when an ack was expected.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ukuhne/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py", line 286, in _wait_for_ask
    ask = self.sp.read(1)[0]
IndexError: list index out of range
). Target didn't respond when an ack was expected.
```

alex-dewar, 2019-07-17:
Some stuff you can try:

* In `software/chipwhisperer/hardware/naeusb/programmer_stm32fserial.py`, find the `writeMemory` method. Try reducing block_size after `except CmdException:` to be much smaller (say 16 bytes, then 8 if that doesn't work).
* In `software/chipwhisperer/hardware/naeusb/serial.py`, find the `write` method. Try increasing the `self.delay()` at the end.

ukuhne, 2019-07-18:
OK, we are getting somewhere. Here is what I did to make it work for me:

1. Changing `block_size` to 16 in `writeMemory()` (in file `programmer_stm32fserial.py`)
2. Same for the functions `readMemory()` and `verifyMemory()`
3. Increase the delay in function `write` (file `serial.py`) to 0.000001 (instead of 0.0000001)

Note that for the block sizes, changing it inside the loop (as you suggested) did not work. It has to be small in the beginning (setting it before the loop) or it will fail even after retrying. Is there a global parameter in the `target` class that could take care of this.

By the way, I noticed that the function `write()` in `serial.py` takes an optional argument `slow` which is not used in the function.

alex-dewar, 2019-07-18:
Good to hear that fixed it for you. There currently isn't any easier way to change `block_size`, but it should be simple enough to add.

I was originally going to use the `slow` argument to make the delay optional, but I don't think it's necessary due to the small length of the delay.

colinoflynn, 2019-07-18:
Alright great! We're going to try and figure out the root cause for this, ideally people aren't having to make so many modifications. Worst case we could at least have a smarter fall-back to small blocks or something, as now that the GUI is not present in the system it's less obvious what you can fiddle with to fix it...

alex-dewar, 2019-07-18:
Hi @ukuhne 

Can you pull the latest commit and try programming without the smaller block size and delay? Increasing the baud rate to 115200 (which is in the latest commit) also fixed the issues on our Linux machine. If you want to try out some other baud rates, you can pass `baud=<value>` to `program_target()`.

colinoflynn, 2019-07-21:
The root cause of this is likely a problem with messages being dropped (see #231). It's easy to confirm the message being dropped in general, and the 'fixes' involving adjusting time delays seem to match up. Will need a firmware fix however it appears. Closing this issue as now tracking in #231.

closed 2019-07-21 by colinoflynn

tresacton, 2019-09-03:
This didn't work for me on STM32F4-HWC.
I receive the same error.

I tried following @ukuhne's advice to reduce the blocksize towards the beginning of each of the 3 functions referenced (first tried 16, then 8).

I couldn't find the delay in function write (file serial.py) so I didn't try changing this - it has been removed in this commit: 30daf375955f33b14b83bb80e27cfa89df2db8ed by @alex-dewar 

Can you please advise of any other things I can try? Would manually setting the baudrate to something else be worth a try? In __init__ it's set to 38400 and in init() it's set to 115200.

tresacton, 2019-09-03:
Just wanted to let you know that I was able to solve this problem but manually re-inserting the code deleted in Alex's commit referenced in above comment, and changing the delay as suggested by ukuhne. Might not be ideal, and it's pretty slow obviously, but I can move forward with programming this target now.

jameshoweee, 2019-10-17:
has this issue been fixed yet? I'm getting exactly the same error..