newaetech/chipwhisperer #230: [cw5dev] scope.default_setup() raises OSError

ukuhne, 2019-07-15:
When I try to setup the scope of my CW light via the python3 API with `scope.default_setup()`, the lights on the board start flashing wildly for about 5 seconds before raising an OSError:

`
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-41-d1f9bd82d6cb> in <module>()
      1 print(scope)
----> 2 scope.default_setup()

/home/xxx/.local/lib/python3.6/site-packages/chipwhisperer-5.0.1-py3.6.egg/chipwhisperer/capture/scopes/OpenADC.py in default_setup(self)
    126 
    127             if count > 100:
--> 128                 raise OSError("Could not lock DCM: {}".format(self))
    129     def dcmTimeout(self):
    130         if self.connectStatus:

OSError: Could not lock DCM: cwlite Device
`

I am running ChipWhisperer under Linux.

alex-dewar, 2019-07-15:
Hi,

Can you try pulling commit b622ecf4d6961d9313488308b402bb0a05016867 and seeing if that fixes it?

ukuhne, 2019-07-16:
This did not fix the error. I tried with two different boards. When I run 
```python
import chipwhisperer as cw
scope = cw.scope()
scope.default_setup()
```
this **always** fails (with the board now flashing slower as probably expected).

Strangely, if I print the scope before the setup:
```python
import chipwhisperer as cw
scope = cw.scope()
print (scope)
scope.default_setup()
```
This succeeds in 4 out of 10 tests. There seems to be some kind of race condition. The loop within the setup function does not seem to help: It either succeeds immediately or not at all. Hope this is useful for debugging.

colinoflynn, 2019-07-16:
I suspect we are seeing issues that were masked with the 'gui interaction code' before (which was called even without the gui). I've made a small change in commit d8b3c41 that reorders the timeout to make more sense, can you see if that changes anything?

It sounds like it may be an invalid DCM setting that will never lock perhaps, so need to see if there is more debug that we can insert.

ukuhne, 2019-07-16:
This fixes the issue for me. Thanks @colinoflynn and @alex-dewar for the quick response!

closed 2019-07-16 by ukuhne

colinoflynn, 2019-07-16:
Great! Thanks for reporting - we're getting close to our initial "Real CW5 Release" so I'm glad to make sure we close off issues like this that would immediately break many peoples machines :)