newaetech/chipwhisperer #231: ChipWhisperer Serial Drops TX Messages if Cannot Keep Up

colinoflynn, 2019-07-21:
When transmitting, ChipWhisperer does not block if the buffers are full on the CW-Lite/CW-Pro. The result is that large messages get dropped - this is likely the root cause of the occasional STM32F programming errors reported in #229 (and seen previously). For #229 increasing the baud rate seemed to fix the problem, suggesting the higher baud rate emptied the buffer quickly enough.

It appears the fix will require an update of the ChipWhisperer firmware, as there is no exposed function to check how full the transmit buffers are. Will test a basic fix (switch to blocking inside USB) but that risks USB timeouts, best would be to have some method of checking the free buffer status.

nils-wisiol, 2020-04-15:
This probably also affects UART messages sent to the target board. My setup is a CW Lite with CW308 Target board and an STM32F UFO. I connected the CW Lite via USB to my computer and use it for programming (works fine) and UART communication with the STM32F (works only for limited message size -- see below).

For debugging, I am running wireshark on my computers USB bus and connected an additional USB/serial converter to the appropriate pins on the UFO board. When I run the following code for `SIZE = 64`, I can see the CW Lite sending the complete message, followed by a line break `\n` to the target. (I can also see the complete message on the USB bus.)

    import chipwhisperer as cw
    scope = cw.scope()
    target = cw.target(scope, cw.targets.SimpleSerial)
    scope.default_setup()
    
    SIZE = 65  # SIZE <= 64 are working well, for SIZE >= 65 messages will be truncated
    key = bytes(list(range(SIZE)))
    text = key

    target.flush()
    target.simpleserial_write('k', key)

However, if I change to `SIZE = 65` in the code above, then I can still see the complete message, including the final line break, on the USB bus, however, on the UART, the line break disappeared. Also, longer messages get truncated in a similar way, while being fully present on the USB bus.

The issue does not occur when I run this through the python debugger, step-by-step debugging the process. It can also be worked around on the host side by [inserting `time.sleep(0.05)` in the loop that sends the data](https://github.com/nils-wisiol/chipwhisperer/commit/e33c5683729078b9b13f42fd6d1ce5388643b687).

alex-dewar, 2020-08-27:
This should be fixed by the newest firmware update (0.20) for the Lite/Nano: 338ee7cb315b4be0c0c6b2843041d01838916318

closed 2020-08-27 by alex-dewar