newaetech/chipwhisperer #233: ReadTheDocs.io link not working

Wmyers559, 2019-07-29:
Hello again,

The nice link in the readme to the [docs](https://chipwhisperer.readthedocs.io/en/stable/getting-started.html) doesn't seem to be working--it shows up with some nice ascii art suggesting that the page doesn't exist. Since changing the `stable` tag to `latest` in the URL gets it to work, it seems that you either need to change the url in the README, or create a stable branch for ReadTheDocs.

Thanks for all your work!

alex-dewar, 2019-07-29:
Thanks for the catch! We were planning on making stable our main ReadtheDocs page, but last minute changes got in the way. Should be fixed by 8ad1863ed06319092d86c719e5638a70570ead4c

closed 2019-07-29 by alex-dewar

Wmyers559, 2019-07-29:
Sounds good! Thanks!