newaetech/chipwhisperer #237: USBC Hub/Adapter Incompatability

alex-dewar, 2019-08-06:
Rankie 2pack USBC high-speed C to USB 3 causes red/blue LEDs near SAM programming port to blink and device to not respond.

jpcrypt, 2019-10-02:
FWIW I have the exact same adapter and what I've observed is that it's picky about which cables it will work with -- some cables which work fine with USB-A ports don't work with this adapter, others do.

tigger1005, 2019-10-17:
I have this issue. I have tried it with two different adapters a usb hub (https://www.charjenpro.com/premiumhub) and a simple usb c to usb-a dongle. Both show the problem. The strange thing is if I repeat trying reconnect and repeat, suddenly it works. So it looks like it is some corner case. Currently I cannot say what to do, or to find a specific procedure. It looks like random, and it take sometime 10 minutes to Brigit to work. (MacBook Pro 2018)