newaetech/chipwhisperer #243: Can't connect the browser

openskypcsc, 2019-09-06:
I installed the CW 5.1.1 VM on a Windows 10 host using VirtualBox (with extensions). The VM boots and runs but I can't seem to get the browser to connect. I have tried many, many network settings and firewall changes but nothing has worked so far.

Tom

colinoflynn, 2019-09-06:
Did the password stuff work (described in https://github.com/newaetech/chipwhisperer/releases/download/5.1.1/Virtual_Machine_CW511_README.pdf)?

Can you check inside the vm with "ps ax | grep jupyter" to check if jupyter-notebook is running?

openskypcsc, 2019-09-07:
Yes I followed the README fie exactly and even tried some variations. I was
able to change both passwords without an issue and when I entered: "jupyter
notebook" I got this response:

[I 01:17:44.113 NotebookApp] The port 8888 is already in use, trying
another port

[I 01:17:44.173 NotebookApp] The Jupyter Notebook is running at:

[I 01:17:44.173 NotebookApp] http://stretch:8889/

[I 01:17:44.173 NotebookApp] Use Control-C to stop this server and shut
down all kernels (twice to skip confirmation).

 I'm a newbie at linux but doesn't that mean Jupyter Notebook is running?

Any suggestions?

Thanks
*Tom*

colinoflynn, 2019-09-09:
Hi Tom,

If you reboot the machine & just check it's running, what output do you get? The port in use suggests it's already running perhaps. It should just start-up all auto-magically.

Do you have any firewalls etc on the host that are different from usual? We've got a native Windows installer as well I can open up.. it was still in testing.

openskypcsc, 2019-09-30:
I have not been able to connect to the ChipWhisperer Jupyter VM running in VirtualBx 6.0 on a Windows 10 host. The attached file contains screenshots of the text below.

- VirtualBox 6.0 shows that port 8888 is used for the Remote Desktop Server in the ChipWhisperer Jupyter VM.
- Performing a netstat on the Windows Host shows port 8888 becomes active in “listening” mode on an unreachable network when the VM is booted up
- After logging in and explicitly running Jupyter notebook - As Jupyter notebook starts it says port 8888 is in use and changes to port 8889 and starts running at http://stretch:8889/
- Running netstat again shows port 8889 is not active but 8888 is.
- I have tried several network configurations of VirtualBox but only 0.0.0.0:8888 shows as active.
- "ChipWhisperer 5 Preview" (Nov 23, 2018) on "https://github.com/newaetech/chipwhisperer/releases" has a note that says it does not work with VirtualBox 6.0. Has this changed or is this note still valid?

[CW Screenshots.pdf](https://github.com/newaetech/chipwhisperer/files/3672556/CW.Screenshots.pdf)
?

openskypcsc, 2019-09-30:
I have not been able to connect with the ChipWhisperer VM running in Virtual Box 6.0 on a Windows 10 host.

- VirtualBox 6.0 shows that port 8888 is used for the Remote Desktop Server in the ChipWhisperer Jupyter VM.
- Performing a netstat on the Windows Host shows port 8888 becomes active in “listening” mode on an unreachable network when the VM is booted up
- After logging in and explicitly running Jupyter notebook - As Jupyter notebook starts it says port 8888 is in use and changes to port 8889 and starts running at http://stretch:8889/
- Running netstat again shows port 8889 is not active.
- I have tried several network configurations of VirtualBox but only 0.0.0.0:8888 shows as active.
- "ChipWhisperer 5 Preview" on https://github.com/newaetech/chipwhisperer/releases has a note that says it does not work with VirtualBox 6.0. Has this changed? 
Screenshots are attached:
[CW Screenshots.pdf](https://github.com/newaetech/chipwhisperer/files/3672620/CW.Screenshots.pdf)

Tom

openskypcsc, 2019-09-30:
I have not been able to connect to the ChipWhisperer Jupyter VM running in VirtualBx 6.0 on a Windows 10 host. The attached file contains screenshots of the text below.

- VirtualBox 6.0 shows that port 8888 is used for the Remote Desktop Server in the ChipWhisperer Jupyter VM.
- Performing a netstat on the Windows Host shows port 8888 becomes active in “listening” mode on an unreachable network when the VM is booted up
- After logging in and explicitly running Jupyter notebook - As Jupyter notebook starts it says port 8888 is in use and changes to port 8889 and starts running at http://stretch:8889/
- Running netstat again shows port 8889 is not active but 8888 is.
- I have tried several network configurations of VirtualBox but only 0.0.0.0:8888 shows as active.
- "ChipWhisperer 5 Preview" (Nov 23, 2018) on "https://github.com/newaetech/chipwhisperer/releases" has a note that says it does not work with VirtualBox 6.0. Has this changed or is this note still valid?

[CW Screenshots.pdf](https://github.com/newaetech/chipwhisperer/files/3672556/CW.Screenshots.pdf)

openskypcsc, 2019-10-02:
I deleted the ChipWhisperer Jupyter VM and reloaded it following the README file carefully this time and it worked auto-magically! I believe I logged into vagrant after the reboot the first time. This time I didn't. I just rebooted and then used the browser to connect and it did.

HackerLlama, 2020-01-23:
Can confirm same symptom with Windows 7.
Virtual.Machine.ChipWhisperer.Jupyter-5-1-3.7z
VirtualBox-6.1.2-135663-Win.exe
Oracle_VM_VirtualBox_Extension_Pack-6.1.2.vbox-extpack

Followed readme as well, logged in vagrant/vagrant, issued 'jupyter notebook password' then sudo reboot. localhost:8888 returns a blank page. Attempting to connect via ssh via localhost:2222 results in an authentication failure.

alex-dewar, 2020-03-26:
Looking through this again, it looks like port 8888 is not available inside the VM (maybe another program is stealing it, or Jupyter is starting twice then closing one?). Everything is based on port forwarding between the VM and Windows, so setting up another port for Jupyter to run on should be fairly simple:

1. In VM settings, select the network tab
1. Hit the "Advanced" drop down
1. Click on port forwarding and add a new port forwarding rule
1. Select a new port for Jupyter to run on (8889 for example), and add these in in "Host Port" and "Guest Port"
1. Inside the VM, run `jupyter notebook --port <PORTNUM>`, where <PORTNUM> is the port you added in the previous part.

alex-dewar, 2021-03-25:
Closing as there's been no activity for a while and I believe the above fix works

closed 2021-03-25 by alex-dewar